clearlinux/clr-boot-manager #25: Scope Solus support

ikeydoherty, 2016-12-12:
Solus requires a solid promise in terms of kernel updates.

..CBM.

Currently we employ two bootloaders, depending on UEFI availability:

 - GRUB2 (MBR/BIOS)
 - goofiboot (UEFI, already implemented in CBM)

In addition, we boot with an initramfs, as Solus has to support mad setups like LVM2/encryption (thankfully handled by the internal initramfs cruft, dracut)

Ideally Solus will be switching to CBM before the next ISO snapshot, making use of the namespace feature in CBM.

N.B: "systemd-class" UEFI, and use of GRUB, are common on many distros in conjunction with an initramfs. It would seem that in implementing this for Solus, I should ensure its relatively agnostic so that others can begin to adopt this too, especially in terms of ESP management.

ikeydoherty, 2016-12-12:
As a Solus specific note:

 - Remove hooks for `grub2` package, switch to `ypkg`
 - Remove hooks for `dracut` package, switch to `ypkg`
 - Remove hooks for `kernel` package, deprecate package entirely.
 
 - Create new fatpackages (no splitting of modules) for kernel types (initially `-lts` for 4.9)
 - Create a simple supporting package with the trigger to run after a kernel file shows up, or a new version of goofiboot, and have them dispatch to `clr-boot-manager`.

Solus kernel upgrade policy is currently "in place", i.e. pulling the rug from under the feet. Thus, if necessary, the first migration may still be allowed to do this for one last time, at which point kernel retention is employed by virtue of using CBM.

- Also needed: Disk space checks within CBM to ensure ESP can contain all possible upgrades, and limit it if necessary to accommodate parallel kernels (OEM systems with teeny ESP)

bryteise, 2016-12-12:
@ikeydoherty Do you care about EFI vars for this too?

ikeydoherty, 2016-12-12:
@bryteise probably should yeah. about time we integrate that eh? :D

bryteise, 2016-12-12:
Scarily enough, yes =).

ikeydoherty, 2017-01-07:
OK so CBM wouldn't need to be modified for "not removing source files" - after further thinking I can insist in the package itself that a source path is permanent, so it's then not removed by the package manager when upgrading. Far better I think.

ikeydoherty, 2017-01-07:
Tracking in Solus: https://docs.google.com/document/d/1IIUUVZhV8myK-qX-yE46mDx6xUpP63tO6o2acOuDncA/edit?usp=sharing

ikeydoherty, 2017-01-08:
In my branch I've now added full initrd support for systemd class bootloaders.
For the sake of completion I'll add them to the syslinux loader too.
Next steps will be to include support for GRUB, /etc/kernel/cmdline, and knowing the LUKS UUID for encrypted roots.

ikeydoherty, 2017-01-09:
First round PR up in https://github.com/ikeydoherty/clr-boot-manager/pull/27

ikeydoherty, 2017-01-10:
Dropped old PR and split the configurable related changes into https://github.com/ikeydoherty/clr-boot-manager/pull/28

Then we can do initramfs etc separately.

ikeydoherty, 2017-01-10:
#28 now merged, we can now focus on initrd, GRUB, and other delicacies of evil.

ikeydoherty, 2017-01-16:
Local coverage pre initrd/crypto changes (to beat):
```
Overall coverage rate:
  lines......: 65.7% (1046 of 1591 lines)
  functions..: 79.2% (84 of 106 functions)
```

ikeydoherty, 2017-01-16:
After the initrd changes:

```
Overall coverage rate:
  lines......: 67.0% (1119 of 1671 lines)
  functions..: 80.4% (90 of 112 functions)
```

ikeydoherty, 2017-01-16:
Crypto changes will require a better approach to probing, as we carelessly assume GPT right now.
New method will require us to store a `CbmDeviceProbe` for root and boot, and store `part_uuid` *and* `uuid`*. In the absence of `part_uuid`, we should always fallback to using `uuid`.

My local work has support for LVM on LUKS.

ikeydoherty, 2017-01-17:
PR#35 is now in so we have initrd support landed for 2.0 (next)

ikeydoherty, 2017-01-24:
Created https://github.com/ikeydoherty/clr-boot-manager/pull/37 for GPT/MBR support along with automatic LUKS configuration

ikeydoherty, 2017-01-26:
#37 now merged

ikeydoherty, 2017-02-10:
Closing to redefine the issue correctly.

closed 2017-02-10 by ikeydoherty