clearlinux/clr-boot-manager #36: Support managed UEFI variables

ikeydoherty, 2017-01-23:
`clr-boot-manager` should retain and maintain the UEFI variables (nvvars) for it's own entry, during boot and upgrades. Specifically a policy must be introduced to allow UEFI variables to be configured from an installation process (i.e. `swupd` from the Clear Linux installer)

ikeydoherty, 2017-01-23:
@bryteise any thoughts on how you want this handled? Mostly thinking about that email earlier today.

bryteise, 2017-01-23:
I'd probably say link off to efivars and implement something along the lines of using a configuration file to decide if the entry gets priority or not. We could add other ister ui hooks for this but it should be coordinated between the two. What other things aside from priority do you think a user would really want to configure?

ikeydoherty, 2017-01-24:
Stable ABI or static link it.. ?

bryteise, 2017-01-24:
Static link.

bryteise, 2018-06-14:
Hrm so this isn't in place how it was planned but since cbm does support configuring its own boot entry via EFI variables today I'd say this is complete. Closing...

closed 2018-06-14 by bryteise