clearlinux/clr-boot-manager #53: Only use PartUUID on a GPT rootfs

ikeydoherty, 2017-03-28:
Scenario:

 - UEFI system
 - GPT disk contains `ESP`
 - `MBR` disk contains `/`

Problem:

 - cbm detects `PartUUID` & UUID`  for GPT converted to MBR
 - cbm writes `root=PARTUUID=` with an unsupported `PartUUID`

Currently, if we find an ESP, we niavely assume GPT. The `probe` code must be modified to determine if the root device is `GPT` or not.  If it is GPT, we should write the PartUUID, otherwise free and NULL the PartUUID and write only the UUID.

lebjortvedt, 2017-03-28:
Is there a workaround for this? My laptop won't boot now and I see that the entry is like this _root=PARTUUID=_ 😢

ikeydoherty, 2017-03-28:
@lebjortvedt do you have an MBR rootfs disk?

lebjortvedt, 2017-03-28:
@ikeydoherty I'm sorry but I'm not sure what you mean. 
I attach a screenshot from Easeus Partition Manager. I run Windows 10 as well on the laptop, and I have one SSD and one spinning disk installed. I have marked the partitions used in Solus with red lines:

[image]

ikeydoherty, 2017-03-28:
Right so your OS is installed on the GPT disk?

lebjortvedt, 2017-03-28:
Correct. The MBR disk is just storage. 
It also seems to me that the UUID in the boot loader is the same as the partition editor sees:

[image]

[image]

ikeydoherty, 2017-03-28:
OK what happens when you try to boot?

lebjortvedt, 2017-03-28:
I get this error:
[image]

But the file looks like it's there:
[image]

ikeydoherty, 2017-03-28:
Is there any way I can see the `\loader\entries\Solus*` file ?

lebjortvedt, 2017-03-28:
title Solus 2017.01.01.0
linux /EFI/com.solus-project/kernel-com.solus-project.lts.4.9.18-10
initrd /EFI/com.solus-project/initrd-com.solus-project.lts.4.9.18-10
options root=PARTUUID=59a2798c-326a-4249-8b02-488d825ba97d quiet ro splash

ikeydoherty, 2017-03-28:
OK so the files on the `linux` and `initrd` lines, they *definitely* exist?

lebjortvedt, 2017-03-28:
According to the previous picture - yes.
I mounted the EFI partition in Windows using the command line, and I can navigate around. However, if I open the EFI/com.solus-project folder and type "dir" _Windows_ gives me an error:

[image]

ikeydoherty, 2017-03-28:
What if you do run chkdsk - is it restorable?

lebjortvedt, 2017-03-28:
Hahaha, now it all went to h*ll 😄 
Chkdsk said that it corrected it, but the _com.solus-project_ is now gone.

Don't know if it is possible to manually create the folder and insert the proper files there?

ikeydoherty, 2017-03-28:
The only way to do this is with  a chroot repair from a livecd - and i think clr-boot-manager is going to throw up because of #54 :/

lebjortvedt, 2017-03-28:
Maybe a stupid question, but will it work to reinstall the OS? The home partition can be spared, so that's not a big deal :)

ikeydoherty, 2017-03-28:
Well - if you're up for reinstalling, I can make an unstable Solus ISO with clr-boot-manager integration if you're willing to test? We need to see how it works out for clr-boot-manager too :)

lebjortvedt, 2017-03-28:
Sure, that'd be great :) 
I suppose I'll wipe the root file system and keep the home partition then.
Too tired to try tonight though 😄

ikeydoherty, 2017-03-28:
That's alright - it'll take me a short while to do the ISO :)

lebjortvedt, 2017-03-28:
Thanks! And thanks for the help so far :)

ikeydoherty, 2017-03-29:
OK I'll be pushing a fix for this shortly - just writing the test suite

lebjortvedt, 2017-03-29:
Awesome :) Will I able to install from USB?

Sendt fra mobiltelefon
Med vennlig hilsen/Best regards
Lars-Eivind Bjørtvedt

ikeydoherty, 2017-03-29:
Yep, once I have it all written up, I'll test it local, then put up a PR. Once I've finished work I'll create a new Solus ISO with the new cbm

ikeydoherty, 2017-03-29:
This behaviour has landed in 1.5.1 - I'll now get this into Solus and a test ISO together

ikeydoherty, 2017-03-29:
Alright spinning the ISO now, will let you know when its uploaded

jamesmshaker, 2017-03-29:
Excellent!!! I'll burn a USB and try it.

James

ikeydoherty, 2017-03-29:
Uploading....

**intensity intensifies**

ikeydoherty, 2017-03-29:
ISO File: https://archive.solus-project.com/.donotdistribute/Solus-Q1-CBM-Validation.iso

```
5301562f26406215762420bf305f1d5c986de0a2dd2fc62b3a114257b7be9e8e  Solus-Q1-CBM-Validation.iso
```

lebjortvedt, 2017-03-29:
Here goes!

Sendt fra mobiltelefon
Med vennlig hilsen/Best regards
Lars-Eivind Bjørtvedt

jamesmshaker, 2017-03-29:
Still no successful boot for me.

James

ikeydoherty, 2017-03-29:
Details..?

jamesmshaker, 2017-03-29:
It still could be my error somewhere along the way.   I use Gparted to
create a swap, /, and efi (512MB fat32) partition on a Nvme ssd.  I put the
dell precision m7510 into uefi mode but enable it to recognize legacy ROMS
(so the live USB will boot).  I boot the live Solus and install.
Everything seems to go ok but when it comes time to reboot and I reboot,
Solus isn't found.

James

ikeydoherty, 2017-03-29:
> but enable it to recognize legacy ROMS (so the live USB will boot)

Uh what? The image is UEFI/Legacy hybrid. It does *not* need legacy support. Sounds like you either booted or wrote it wrong.

jamesmshaker, 2017-03-29:
When I use rufus to try and write the solus iso to the usb stick and select
the GPT uefi optionI get this error:

When using UEFI Target Type, only EFI bootable ISO images are supported.
Please select an EFI bootable ISO or set the Tartget type to BIOS.

ikeydoherty, 2017-03-29:
..? You need to write the image *as is*, unmodified. Don't let rufus butcher it.

JoshStrobl, 2017-03-29:
Choose the DD Image Mode as per [our documentation](https://solus-project.com/articles/installation/preparing-to-install/en/#usb-3).

lebjortvedt, 2017-03-29:
It worked for me :+1: 
I reinstalled the root file system but kept the home partition, and that worked. 
The first time I tried to log in everything locked up - but the next times it's worked flawlessly.
The only difference now is that the (dual)boot menu is skipped unless I press the e-key.

ikeydoherty, 2017-03-29:
```
sudo clr-boot-manager set-timeout some-timeout-value
sudo clr-boot-manager update
```

lebjortvedt, 2017-03-29:
Worked like a charm!
Thanks a lot for everything 😀

ikeydoherty, 2017-03-29:
Awesome thanks! I'll close this issue now then

closed 2017-03-29 by ikeydoherty