clearlinux/clr-boot-manager #61: Does not work on Btrfs

silkeh, 2017-03-30:
Running `clr-boot-manager` in Solus installed on a Btrfs filesystem results in the following:

```
# CBM_DEBUG=1 clr-boot-manager update
[INFO] cbm (src/bootman/bootman.c:L437): Current running kernel: 4.9.18-10.lts
[ERROR] cbm (src/lib/files.c:L164): Invalid block device: /
[ERROR] cbm (src/lib/system_stub.c:L30): Invalid block device: 0:34
Out of memory[DEBUG] cbm (src/bootman/bootman.c:L124): UEFI boot now selected (goofiboot)
[INFO] cbm (src/bootman/update.c:L75): Checking for mounted boot dir
[INFO] cbm (src/bootman/update.c:L78): boot_dir is already mounted: /boot
[DEBUG] cbm (src/bootman/update.c:L237): Now beginning update_native
[DEBUG] cbm (src/bootman/update.c:L246): update_native: 1 available kernels
[DEBUG] cbm (src/bootman/update.c:L266): update_native: Running kernel is (lts) ///usr/lib/kernel/com.solus-project.lts.4.9.18-10
[FATAL] cbm (src/bootman/sysconfig.c:L126): sysconfig insane: Missing root device
[FATAL] cbm (src/bootman/update.c:L455): Failed to update bootloader
```

Notes: 
 - It doesn't matter if the `/` filesystem is in the top-level subvolume (`subvol=/`) or another subvolume (eg: `subvol=solus`).
- The block device `0:34` doesn't exist.

ikeydoherty, 2017-05-12:
Will attempt this for the 1.6 release

maxwellsayles, 2019-03-30:
Any update on this?  This is essentially still an issue on Solus 4.0 with clr-boot-manager 3.1.0.

```
$ sudo CBM_DEBUG=1 clr-boot-manager update
[INFO] cbm (../src/bootman/bootman.c:L482): Current running kernel: 4.20.16-112.current
[INFO] cbm (../src/bootman/sysconfig.c:L98): Discovered UEFI ESP: /dev/disk/by-partuuid/f1de6b38-3ec8-4ea2-a8ee-4f09f89bb6e9
[INFO] cbm (../src/bootman/sysconfig.c:L123): Fully resolved boot device: /dev/nvme0n1p1
[ERROR] cbm (../src/lib/system_stub.c:L31): Invalid block device: 0:41
Out of memory[DEBUG] cbm (../src/bootman/bootman.c:L130): UEFI boot now selected (systemd)
[INFO] cbm (../src/bootman/bootman.c:L502): path ///usr/lib/initrd.d does not exist
[INFO] cbm (../src/bootman/update.c:L116): Checking for mounted boot dir
[INFO] cbm (../src/bootman/update.c:L119): boot_dir is already mounted: /boot
[DEBUG] cbm (../src/bootman/update.c:L291): Now beginning update_native
[DEBUG] cbm (../src/bootman/update.c:L300): update_native: 1 available kernels
[DEBUG] cbm (../src/bootman/update.c:L320): update_native: Running kernel is (current) ///usr/lib/kernel/com.solus-project.current.4.20.16-112
[SUCCESS] cbm (../src/bootman/update.c:L335): update_native: Bootloader updated
[FATAL] cbm (../src/bootman/sysconfig.c:L140): sysconfig insane: Missing root device
[ERROR] cbm (../src/bootman/update.c:L346): Failed to repair running kernel
[DEBUG] cbm (../src/bootman/update.c:L358): update_native: Checking kernels for type current
[INFO] cbm (../src/bootman/update.c:L373): update_native: Default kernel for type current is ///usr/lib/kernel/com.solus-project.current.4.20.16-112
[FATAL] cbm (../src/bootman/sysconfig.c:L140): sysconfig insane: Missing root device
[FATAL] cbm (../src/bootman/update.c:L380): Failed to install default-current kernel: ///usr/lib/kernel/com.solus-project.current.4.20.16-112
```

mdhorn, 2019-05-01:
What are the file system formats restrictions for the root '/' parition?
We still see btrfs failing in current leases of Clear Linux OS.
@gtkramer FYI this bug has been around a while.

mcku, 2019-06-06:
Although there is a root in cmdline, the clr-boot-manager attempts to convert "/" into a device using `stat`, which is known to report incorrect device ids (maybe due to a kernel bug). 
Why not parse the cmdline `root` instead?

```
$ cat /proc/cmdline 
root=/dev/nvme0n1p4 quiet ........

$ file /dev/nvme0n1p4    
/dev/nvme0n1p4: block special (259/4)

$ file /dev/block/259\:4 
/dev/block/259:4: symbolic link to ../nvme0n1p4
``` 
However, the cbm tool thinks major, minor to be 0 and 24 respectively.

I tried to trace a little bit and it looks like when it comes to convert "/" path to a device (https://github.com/clearlinux/clr-boot-manager/blob/v3.1.0/src/bootman/sysconfig.c#L128), the `stat` call returns an invalid pair of major and minor numbers. This looks like it may be an issue within the kernel. See: https://stackoverflow.com/questions/39494077/linux-stat2-call-gives-non-existing-device-id

mcku, 2019-06-06:
Also, maybe it helps to put another pointer... `btrfs` userspace tool can correctly identify the physical device as such:
```
$ sudo btrfs filesystem show /
Label: none  uuid: ....................
	Total devices 1 FS bytes used 214.35GiB
	devid    1 size 400.00GiB used 389.32GiB path /dev/nvme0n1p4
```
So, the approach there might be utilized if found appropriate. Here is the pointer to its code:
https://github.com/kdave/btrfs-progs/blob/43013422dbce4bcc9ed77cfe65b294caa0985ec8/cmds-filesystem.c#L676

bryteise, 2019-06-06:
@mcku thanks for the insights and pointers =).

themagicalmammal, 2021-02-07:
Does this issue has any workaround? I am using Btrfs using(Btrfs-convert) but the kernel doesn't update at all.

silkeh, 2021-02-07:
@themagicalmammal you can manually build clr-boot-manager with the patch from #238 ([here](https://github.com/clearlinux/clr-boot-manager/pull/238.patch)). What Distro?

themagicalmammal, 2021-06-22:
> @themagicalmammal you can manually build clr-boot-manager with the patch from #238 ([here](https://github.com/clearlinux/clr-boot-manager/pull/238.patch)). What Distro?

Clear itself

PR #238 (2021-06-30): Add support for non-ext filesystems
Calling stat() on some filesystems, like Btrfs,
results in a dev_t with no relation to the backing device(s),
and subsequent failure to detect the root fs.

Add support such filesystems by retrieving the block device from
`/proc/self/mounts` and working on from there.

Resolves #61, #182 and #193.
commit 74dd4636 (2021-06-30): Add support for non-ext filesystems

Calling stat() on some filesystems, like Btrfs,
results in a dev_t with no relation to the backing device(s),
and subsequent failure to detect the root fs.

Add support such filesystems by retrieving the block device from
`/proc/self/mounts` and working on from there.

Resolves #61, #182 and #193.
closed 2021-06-30 by fenrus75