clearlinux/clr-boot-manager #65: Multiple Kernel Versions at Goofiboot

tayspen72, 2017-04-04:
With the update of the clr-boot-manager, the goofiboot is now showing multiple kernel versions as possible boot options. I have ran the clr-boot-manager report-booted and update but the current and previous options remain. I would like to remove the previous option and old kernel data stored but cannot find information on how to do so under the manpages for clr-boot-manager. Sugguestions?

ikeydoherty, 2017-04-04:
Are they all CBM managed kernels? It could be an artefact of the transition:

```
ls -lRa /boot
ls -lRa /usr/lib/kernel
```

tayspen72, 2017-04-04:
Here is the output. I am dual booted with windows, as this obviously shows.

$ ls -lRa /boot
/boot:
total 38408
drwxr-xr-x  3 root root     4096 Apr  3 11:14 .
drwxr-xr-x 17 root root     4096 Apr  3 11:10 ..
lrwxrwxrwx  1 root root        1 Apr  3 05:03 boot -> .
drwxr-xr-x  6 root root     4096 Dec 31  1969 efi
-rw-------  1 root root 19650869 Jan  1 13:17 initramfs-4.8.15.img
-rw-------  1 root root 19664245 Apr  3 11:14 initramfs-4.9.20-12.lts.img

/boot/efi:
total 24
drwxr-xr-x 6 root root 4096 Dec 31  1969  .
drwxr-xr-x 3 root root 4096 Apr  3 11:14  ..
drwxr-xr-x 6 root root 4096 Mar 27 10:35  EFI
drwxr-xr-x 3 root root 4096 Apr  4 07:59  loader
drwxr-xr-x 2 root root 4096 Oct 23 13:26 'System Volume Information'
drwxr-xr-x 4 root root 4096 Oct 14 01:22  .Trash-1000

/boot/efi/EFI:
total 24
drwxr-xr-x 6 root root 4096 Mar 27 10:35 .
drwxr-xr-x 6 root root 4096 Dec 31  1969 ..
drwxr-xr-x 2 root root 4096 Apr  3 11:07 Boot
drwxr-xr-x 2 root root 4096 Apr  2 09:27 com.solus-project
drwxr-xr-x 2 root root 4096 Apr  3 11:07 goofiboot
drwxr-xr-x 4 root root 4096 Mar 29  2016 Microsoft

/boot/efi/EFI/Boot:
total 92
drwxr-xr-x 2 root root  4096 Apr  3 11:07 .
drwxr-xr-x 6 root root  4096 Mar 27 10:35 ..
-rwxr-xr-x 1 root root 85227 Oct 18 00:00 BOOTX64.EFI

/boot/efi/EFI/com.solus-project:
total 52872
drwxr-xr-x 2 root root     4096 Apr  2 09:27 .
drwxr-xr-x 6 root root     4096 Mar 27 10:35 ..
-rwxr-xr-x 1 root root 19973450 Mar 27 10:35 initrd-com.solus-project.lts.4.9.18-10
-rwxr-xr-x 1 root root 20343328 Apr  2 09:27 initrd-com.solus-project.lts.4.9.20-12
-rwxr-xr-x 1 root root  6903376 Mar 27 10:35 kernel-com.solus-project.lts.4.9.18-10
-rwxr-xr-x 1 root root  6903376 Apr  2 09:27 kernel-com.solus-project.lts.4.9.20-12

/boot/efi/EFI/goofiboot:
total 92
drwxr-xr-x 2 root root  4096 Apr  3 11:07 .
drwxr-xr-x 6 root root  4096 Mar 27 10:35 ..
-rwxr-xr-x 1 root root 85227 Oct 18 00:00 goofibootx64.efi

/boot/efi/EFI/Microsoft:
total 20
drwxr-xr-x  4 root root 4096 Mar 29  2016 .
drwxr-xr-x  6 root root 4096 Mar 27 10:35 ..
drwxr-xr-x 42 root root 8192 Mar 29  2016 Boot
drwxr-xr-x  2 root root 4096 Mar 29  2016 Recovery

/boot/efi/EFI/Microsoft/Boot:
total 4708
drwxr-xr-x 42 root root    8192 Mar 29  2016 .
drwxr-xr-x  4 root root    4096 Mar 29  2016 ..
-rwxr-xr-x  1 root root   45056 Apr  3 18:51 BCD
-rwxr-xr-x  1 root root   65536 Mar 29  2016 BCD.LOG
-rwxr-xr-x  1 root root       0 Mar 29  2016 BCD.LOG1
-rwxr-xr-x  1 root root       0 Mar 29  2016 BCD.LOG2
drwxr-xr-x  2 root root    4096 Oct 23 13:26 bg-BG
-rwxr-xr-x  1 root root 1193824 Mar  4 00:20 bootmgfw.efi
-rwxr-xr-x  1 root root 1178976 Mar  4 00:20 bootmgr.efi
-rwxr-xr-x  1 root root   65536 Oct 23 13:26 BOOTSTAT.DAT
[... 418 lines cut ...]
drwxr-xr-x 2 root root 4096 Oct 14 01:22 .
drwxr-xr-x 4 root root 4096 Oct 14 01:22 ..
-rwxr-xr-x 1 root root   72 Oct 14 01:22 bootx64.efi.trashinfo

$ ls -lRa /usr/lib/kernel
/usr/lib/kernel:
total 30312
drwxr-xr-x   3 root root     4096 Apr  3 11:13 .
drwxr-xr-x 156 root root    69632 Apr  4 11:00 ..
-rw-r--r--   1 root root       15 Mar 31 13:58 cmdline-4.9.20-12.lts
-rwxr-xr-x   1 root root  6903376 Mar 31 13:58 com.solus-project.lts.4.9.20-12
-rw-r--r--   1 root root   177768 Mar 31 13:58 config-4.9.20-12.lts
lrwxrwxrwx   1 root root       31 Apr  3 11:13 default-lts -> com.solus-project.lts.4.9.20-12
-rw-------   1 root root 20343328 Mar 31 13:58 initrd-com.solus-project.lts.4.9.20-12
drwxr-xr-x   2 root root     4096 Apr  3 11:14 install.d
-rw-r--r--   1 root root  3522632 Mar 31 13:58 System.map-4.9.20-12.lts

/usr/lib/kernel/install.d:
total 24
drwxr-xr-x 2 root root 4096 Apr  3 11:14 .
drwxr-xr-x 3 root root 4096 Apr  3 11:13 ..
-rwxr-xr-x 1 root root  191 Mar 28 20:05 50-depmod.install
-rwxr-xr-x 1 root root 1242 Mar 31 16:32 50-dracut.install
-rwxr-xr-x 1 root root 2263 Mar 31 16:32 51-dracut-rescue.install
-rwxr-xr-x 1 root root 2040 Mar 28 20:05 90-loaderentry.install

ikeydoherty, 2017-04-04:
OK so in a fully booted system, when running the `4.9.20`, just run:

`sudo clr-boot-manager update`

Note that you're always going to have multiple kernel versions going forward, thats how CBM works. The idea is that when you get the *next* update, the previous to current is removed.

tayspen72, 2017-04-04:
OK. I have ran that command but you're saying I can just count on the next kernel update to remove the previous kernels. Thanks for the quick response. Very much appreciated

closed 2017-04-04 by tayspen72

ikeydoherty, 2017-04-05:
np. What we do for UEFI is actually set the default kernel entry. So if you're using `linux-lts`, when you reboot it'll automatically select the new kernel each time, meaning you don't need to do anything yourself :)