clearlinux/clr-boot-manager #68: Add a symlink for the current initrd?

ddulic, 2017-04-30:
Relevant: https://solus-project.com/forums/viewtopic.php?f=4&t=7211&start=20#p21727

Just a symlink to the current initrd, just like it exists for default-lts.

Possible?

Thanks.

ikeydoherty, 2017-04-30:
I'm not sure I follow, we already create `/vmlinuz` and `/initrd.img` symlinks when using GRUB.
The files in `/usr/lib/kernel` are *staging* files and should *never* be detected by GRUB. Boot files only go into `/boot`.

Also I would like to point out that the thread in question lacks honesty/details:

 - Solus cannot be installed on GPT unless it is in in UEFI mode. It's forbidden in the installer.
 - clr-boot-manager only creates GRUB links (`/vmlinuz` and `/initrd.img`) when using the GRUB bootloader, and legacy boot.

What I actually see in the bug report is circumvention of the boot and installation mechanisms, which I'm sad to say is entirely unsupported.

closed 2017-04-30 by ikeydoherty

ikeydoherty, 2017-04-30:
Also I've tried to see a way around this, but honestly I can't. If a separate /boot is being used, other
GRUBs won't be able to work out what the links actually point to or how to boot them..

ddulic, 2017-04-30:
Thanks, did not know a symlink to a initrd wouldn't work.

ikeydoherty, 2017-04-30:
Yeah the files are in `/` for links :)