clearlinux/clr-boot-manager #77: Need to integrate firmware capsule updates

ikeydoherty, 2017-10-06:
In Solus we're trying to ensure all aspects of the users system is up to date, to that extent we:

 - Early-load microcode updates for the CPU
 - Ship recent firmware and kernels
 - Provide rolling stack updates for critical software

However we currently DO NOT have firmware updates. clr-boot-manager is the obvious integration
point in the deployment chain, and should be able to manage `fwupd` or similar to enable the deployment
of FW capsule updates for our UEFI users.

It also makes sense to integrate the points here as CBM is responsible for boot items, and could easily change the "NextBoot" to ensure we reboot to the firmware update screen.

..Discuss? :)

bryteise, 2017-10-07:
fwupd is definitely something we are interested in getting integrated into cbm. We need to do uefi var management in cbm anyway so next boot like things should definitely be in scope.

Do you see fwupd providing a library interface so that cbm can use it and then fwupd having a cli interface as well?

I imagine most of the discussion will be around how fwupd will operate. Ideally cbm would consume it in fairly straightforward manor as there shouldn't be too many fancy things other than detect if new firmware is available (and cbm then switching the next boot). Does that seem about right?

busykai, 2017-10-08:
@ikeydoherty, do you mean setting EFI `BootNext` var? is `fwupd` a boot service? or do you mean starting a linux into `fwupd` screen? (I need to familiarize myself more with `fwupd`.)

Just to call your attention to #75 which contains an API for EFI var manipulation (`BootXXXX` for now), but it would be well extended to work with other boot variables too. I'll be finishing up that PR next week, not much work left there..

ikeydoherty, 2017-10-18:
@busykai yeah this stuff depends on your efi var stuff I reckon :)

alecbcs, 2018-07-09:
Hi everyone, has there been any progress on this issue since October?

Jacalz, 2018-08-06:
It would be really nice seeing this included 👍

Vistaus, 2018-08-06:
I just read that Lenovo has joined the LVFS (https://phoronix.com/scan.php?page=news_item&px=Lenovo-LVFS-Support). Now I can easily update the firmware of my 2018 ThinkPad. But as I use Solus, I need to wait for this to be resolved. So it would be great if you guys could integrate this :)

Jacalz, 2018-12-07:
Any new on getting this integrated? It would be such a great feature for a lot of people and it would be a safe and smart integration of it compared to many other ones in the Linux universe. Can barely wait until we get this integrated 😇

alecbcs, 2019-05-07:
Any updates on getting this integrated? @bryteise how much work would need to be done on this issue to implement LVFS? Thank you!

bryteise, 2019-05-07:
@alecbcs At this point because fwupdate is available I'm mostly in favor of closing this issue as I don't think integrating fwupd with CBM is the way to go.

alecbcs, 2019-05-07:
@bryteise From what I understand fwupdate queues the updates on the `/boot` partition. However, that means right now the user has to manually mount the `/boot` partition when they want to update BIOS/firmware. Do you know of anyway we would automate this process? It also looks like `fwupd` recently integrated the code from `fwupdate` so fwupdate might not be needed anymore in this process if we can do the same thing with `fwupd`. Do you know from the systems side what we might have to do to give `fwupd` access to, in order to mount the boot partition?

bryteise, 2019-05-07:
@alecbcs Yes, fwupd is the user space integration that manages the overall update process and contains fwupdate. fwupd doesn't really do well with mounting /boot itself but the systemd service for fwupd could be modified to have it mount the /boot partition as part of it running.

alecbcs, 2019-05-07:
@bryteise Okay interesting! If all that needed, is to mount the /boot partition before running fwupd, maybe we could consider having the package manager mount the /boot partition before calling `fwupdmgr update`?

bryteise, 2019-05-09:
fwupd really wants to run as an active service that is constantly looking for updates but it isn't super friendly still in my opinion so I would be more likely to advise the system administer to do control its running and what it is trying to update.

dorileo, 2020-01-30:
I agree we don't want cbm overlapping with fwupd. I would close this issue.