clearlinux/clr-boot-manager #84: Tracking: Make Solus happy again

ikeydoherty, 2017-10-31:
Tracking some issues here as I'll need to follow up with some PRs to restore git master functionality.

Currently tests will fail UNLESS `--with-bootloader=shim-systemd-boot`. In Travis this passes only because the `auto` mode will fallback to `shim-systemd-boot`. All other options will fail on `check_uefi`.

There is also the issue of files having moved into `kernel/` subdirectory, so I'll need to track migration paths.

AR ikey:

 - [ ] Make `--with-bootloader=goofiboot` work again (and for all sd class.)
 - [ ] Prep Travis PR to build in multiple configurations (shim and systemd-class)
 - [ ] Add migration path for `kernel/` tree moving

General:

 - [ ] `distcheck` is broken on master (William found this) - cmdline tests no longer pass. LIkely related to the above issue.
 - [ ] Need to update documentation around new shim system
 - [ ] Configure doesn't check for `gnu-efi-devel` and `efivar-devel` (Solus names but yknow the dudes I mean)

ikeydoherty, 2017-11-01:
Related: https://github.com/clearlinux/clr-boot-manager/pull/87

ikeydoherty, 2017-11-01:
Much of this now fixed by merging #87 in cba2f74c8c4704b24ff1f0921dae906cc416842f

bryteise, 2017-12-13:
So there isn't any need to do the migration path now correct? Still need to update the shim system though X(.

ikeydoherty, 2017-12-15:
Ah right yeah this issue itself can prolly be closed or edited. What we actually need is EFI var management unbound from shim ^^

dorileo, 2020-01-30:
Well, I'm closing this issue. Please feel free to reopen it or create new issues for features we've not implemented yet.

Thanks...

closed 2020-01-30 by dorileo