clearlinux/clr-boot-manager #103: Format of cmdline.d/*.conf should be specified

bwarden, 2018-03-15:
https://github.com/clearlinux/clr-boot-manager/blob/50cc4ea5f967dcf7a51a4c089d4f7bb4cfae9160/man/clr-boot-manager.1.in#L101

We should specify that the contents of any cmdline.d/*.conf must be simple text to be appended directly to the kernel command line.

We should also mention that contents from /usr/lib/kernel/cmdline* and /usr/lib/kernel/cmdline.d/*.conf are used in generation of the command line.

bryteise, 2018-09-12:
@bwarden Mind taking a look at #127 to see if this is what you were hoping to see?

dorileo, 2019-12-13:
@bwarden  it seems this issue was fixed with #127. Please feel free to reopen it if think we're still missing something.

Thanks.

closed 2019-12-13 by dorileo