clearlinux/clr-boot-manager #109: Feature request: enable selection of kernels in next boot on command line

xulongwu4, 2018-05-05:
I am a Solus user which uses clr-boot-manager for boot management. The user is expected to select the linux kernel to boot up during start up by spamming the space key (on UEFI hardware).

I have a laptop with UEFI setup. However, I cannot select the boot kernels by the above approach. My problem is that, the laptop becomes unresponsive once I hit the space key during startup and enter the page for selection of kernels. I can see which kernels are available but I cannot select any of them. Even worse, the laptop would freeze there with no response to any keystroke. It boots fine if I let it boot regularly to the log in page.

Because of this issue, I cannot switch the kernels on this laptop. I am proposing to add a command line option to clr-boot-manager so that I can choose what kernel to use in the next bootup. If it works fine, then I still can use `clr-boot-manager report-booted` to keep this option. Otherwise the option would not be saved and in future startups the machine will use the original kernel it used. In addition, there can an option to list all available kernels.

One example of such a feature could be:

`clr-boot-manager list-kernels -- list all kernels installed on the machine`
`clr-boot-manager select-kernel XXX -- select a kernel for next boot`

bryteise, 2018-05-05:
So I think you can work around the issue for now with clr-boot-manager set-timeout but I do think we should add this feature.

xulongwu4, 2018-08-04:
@bryteise The 'set-timeout' workaround does not work for me since the system freezes upon any keystroke at the boot menu. When I use a timeout of '5' seconds, I can see the boot menu count down. But I still have no way to select any boot entry to boot.

bryteise, 2018-08-11:
@xulongwu4 Ugh okay, well probably efibootmgr is your short term option but your example use would be good to support.

bryteise, 2018-11-14:
Fixed in #259

closed 2018-11-14 by bryteise