clearlinux/clr-boot-manager #111: Retain `loader.conf` settings

ahkok, 2018-05-21:
It should be possible to retain settings other than `default` and `timeout` in the `loader.conf`. Specifically, the 4 following options should be retained at all times:

* `editor`
* `auto-entries`
* `auto-firmware`
* `console-mode`

right now these are wiped every time clr-boot-manager runs.

cunidev, 2018-06-06:
Definitely agree. My Windows partition can't be accessed if the timeout is reset to 0 (and the shift menu trigger has some crazy short timeout). It'd also be nice to set things like timeout directly via clr-boot-manager without manually accessing the ESP, though that's another thing.

bryteise, 2018-06-06:
@cunidev I might be misunderstanding you but isn't clr-boot-manager set-timeout && clr-boot-manager update doing just that?