clearlinux/clr-boot-manager #120: [FATAL] cbm: (../src/bootman/update.c:L471): Failed to install bootloader

rebeccasf, 2018-07-26:
Environment: Bare metal installation of Clear
Last working Clear Linux version: 23690
Current Clear Linux version installed: 23950

Trying to set timeout for systemd-boot menu and perform clr-boot-manager update.

$ sudo clr-boot-manager update
[FATAL] cbm: (../src/lib/bootvar.c:L411): efi_generate_file_device_path_from_esp() failed: Function not implemented
[FATAL] cbm: (../src/lib/bootvar.c:L411): efi_generate_file_device_path_from_esp() failed: Function not implemented
[FATAL] cbm: (../src/bootloader/shim-systemd.c:L283): Cannot create EFI variable (boot entry)
[FATAL] cbm: (../src/bootman/update.c:L471): Failed to install bootloader

The timeout is set by system did not receive the update hence the boot menu does not show up when system reboot.

busykai, 2018-07-26:
@rebeccasf, what's the hardware you ran this on? also, could you please provide output of `sudo fdisk -l`?

rebeccasf, 2018-07-26:
Output:

Disk /dev/mmcblk0: 58.2 GiB, 62537072640 bytes, 122142720 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 1048576 bytes
Disklabel type: gpt
Disk identifier: 6EACB724-8677-4212-B8B6-EB8D6D81CD5E
 
Device           Start       End   Sectors  Size Type
/dev/mmcblk0p1    2048   1048575   1046528  511M EFI System
/dev/mmcblk0p2 1048576   1114111     65536   32M Linux swap
/dev/mmcblk0p3 1114112 122140671 121026560 57.7G Linux root (x86-64)
 
 
Disk /dev/mmcblk0boot1: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 1048576 bytes
 
 
Disk /dev/mmcblk0boot0: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 1048576 bytes

busykai, 2018-07-26:
Could you try `modprobe efivars` and then running `clr-boot-manager update` again?

miguelinux, 2018-07-26:
@bryteise

This is the same issue as Up Board 2 (UP2)

The issue is that the efivar package was updated but has some changes which broke CBM

bryteise, 2018-07-26:
@miguelinux Yea, likely the same issue since this is ENOSYS. @busykai if you look at https://github.com/rhboot/efivar/blob/master/src/linux.c#L498, you'll see where the suspected problem is. For some reason the interface type is not being identified and efivar v36 now errors out in those cases.

rebeccasf, 2018-07-27:
@busykai, i still see the same error after modprobe. efivars does not show up in the module list in lsmod.

busykai, 2018-07-27:
@rebeccasf, i'm going to close this issue. the root cause is in efivar, fix submitted (rhboot/efivar#116), clr-boot-manager (and swupd) should be functional with the next Clear release.

closed 2018-07-27 by busykai