clearlinux/clr-boot-manager #123: clr-boot-manager can't remove entries like intel_iommu=igfx_off

eadamsintel, 2018-08-14:
The default clear linux boot options has intel_iommu=igfx_off which can't be overridden with intel_iommu=on.  Using the standard method of placing a conf file in /etc/kernel/cmdline.d and running clr-boot-manager update to add the intel_iommu=on setting does add the setting but that setting does not override the igfx_off setting.  Therefore clr-boot-manager needs a way to remove kernel entries and/or overwrite them.

bryteise, 2018-09-06:
Will be in the next release.

closed 2018-09-06 by bryteise

jbhanks, 2018-10-21:
I hope this isn't hijacking the thread too much, but what should I change that option to if I am creating a Clear Linux VM on an AMD CPU?

electrocucaracha, 2019-04-18:
@bryteise I'm using the latest version of the `clr-boot-manager` and it seems like the cmdline removal feature is not working.

```
$ clr-boot-manager version
clr-boot-manager - version 3.1.0
$ cat /proc/cmdline 
root=PARTUUID=a8c44c65-c1cb-4f00-9e82-7adab82cc074 quiet console=tty0 console=ttyS0,115200n8 init=/usr/bin/initra-desktop  initcall_debug tsc=reliable no_timer_check noreplace-smp kvm-intel.nested=1 rootfstype=ext4,btrfs,xfs intel_iommu=igfx_off cryptomgr.notests rcupdate.rcu_expedited=1 rcu_nocbs=0-64 rw
$ echo "intel_iommu=igfx_off" | sudo tee /etc/kernel/cmdline-removal.d/disable-iommu.conf
$ sudo clr-boot-manager update
$ sudo reboot
```

After the reboot:

```$ cat /proc/cmdline
root=PARTUUID=a8c44c65-c1cb-4f00-9e82-7adab82cc074 quiet console=tty0 console=ttyS0,115200n8 init=/usr/bin/initra-desktop  initcall_debug tsc=reliable no_timer_check noreplace-smp kvm-intel.nested=1 rootfstype=ext4,btrfs,xfs intel_iommu=igfx_off cryptomgr.notests rcupdate.rcu_expedited=1 rcu_nocbs=0-64 rw```

bryteise, 2019-05-01:
Yep, there is a bug where this is not working correctly atm.

electrocucaracha, 2019-05-20:
@bryteise is possible to reopen this issue until it's fixed?

bryteise, 2019-05-20:
@electrocucaracha Ah the fix is in and tested in the latest Clear Linux release. I just merged the fix PR into git (#151).