clearlinux/clr-boot-manager #132: Clear AWS image doesn't update kernel properly

bktan8, 2018-12-04:
When I updated my Clear AWS instance to a newer kernel, the kernel file was installed fine, but got a CBM error and /boot/syslinux.cfg was still configured for the older kernel. I had to manually edit syslinux.cfg to point to the new kernel.

bryteise, 2018-12-04:
What was the CBM error?

bktan8, 2018-12-04:
[FATAL] cbm (../src/bootman/update.c:L393): Failed to install bootloader

----log----
> Un-updated version of Clear
clear@clr-d43af41c51a5449ab6f5e76a00c3465e ~ $ swupd info
Installed version: 26520
Version URL:       https://cdn.download.clearlinux.org/update/
Content URL:       https://cdn.download.clearlinux.org/update/

> Un-updated kernel 
clear@clr-d43af41c51a5449ab6f5e76a00c3465e ~ $ uname -a
Linux clr-d43af41c51a5449ab6f5e76a00c3465e 4.18.16-80.aws #2 SMP Sun Oct 21 14:56:15 UTC 2018 x86_64 GNU/Linux

> Updating...
clear@clr-d43af41c51a5449ab6f5e76a00c3465e ~ $ sudo swupd update
Update started.
Preparing to update from 26520 to 26670
Downloading packs...

Extracting kernel-aws pack for version 26590
	...100%

Statistics for going from version 26520 to version 26670:

    changed bundles   : 3
    new bundles       : 0
    deleted bundles   : 0

    changed files     : 171
    new files         : 634
    deleted files     : 635

Starting download of remaining update content. This may take a while...
	...100%
Finishing download of update content...
Staging file content
Applying update
	...100%
Update was applied.
Calling post-update helper scripts.
none
[FATAL] cbm (../src/bootman/update.c:L393): Failed to install bootloader
rngd.service: needs a restart (a library dependency was updated)
pacdiscovery.service: needs a restart (the binary was updated)
tallow.service: needs a restart (a library dependency was updated)
systemd-udevd.service: needs a restart (a library dependency was updated)
pacrunner.service: needs a restart (the binary was updated)
systemd-journald.service: needs a restart (a library dependency was updated)
systemd-resolved.service: needs a restart (a library dependency was updated)
systemd-timesyncd.service: needs a restart (a library dependency was updated)
Update took 20.1 seconds
182 files were not in a pack
Update successful. System updated from version 26520 to version 26670

> New Clear version
clear@clr-d43af41c51a5449ab6f5e76a00c3465e ~ $ swupd info
Installed version: 26670
Version URL:       https://cdn.download.clearlinux.org/update/
Content URL:       https://cdn.download.clearlinux.org/update/

> reboot and log back in

> Kernel version 
clear@clr-d43af41c51a5449ab6f5e76a00c3465e ~ $ uname -a
Linux clr-d43af41c51a5449ab6f5e76a00c3465e 4.18.16-80.aws #2 SMP Sun Oct 21 14:56:15 UTC 2018 x86_64 GNU/Linux

> Was expecting org.clearlinux.aws.4.18.16-81 to show up here, but no.
clear@clr-d43af41c51a5449ab6f5e76a00c3465e ~ $ ls -l /boot
total 7564
-r--r--r-- 1 root root  115604 Nov 27 14:17 ldlinux.c32
-r--r--r-- 1 root root   60928 Nov 27 14:17 ldlinux.sys
-rw-r--r-- 1 root root 7557168 Nov 27 14:17 org.clearlinux.aws.4.18.16-80
-rw-r--r-- 1 root root     358 Nov 27 14:17 syslinux.cfg

> Copy of new kernel org.clearlinux.aws.4.18.16-81 is found in /usr/lib/kernel
clear@clr-d43af41c51a5449ab6f5e76a00c3465e ~ $ ls -al /usr/lib/kernel
total 14924
drwxr-xr-x  2 root root    4096 Dec  4 22:38 .
drwxr-xr-x 12 root root    4096 Dec  4 22:38 ..
-rw-r--r--  1 root root     186 Oct 21 14:56 cmdline-4.18.16-80.aws
-rw-r--r--  1 root root     186 Nov 30 01:23 cmdline-4.18.16-81.aws
-rw-r--r--  1 root root   89065 Oct 21 14:56 config-4.18.16-80.aws
-rw-r--r--  1 root root   89547 Dec  4 22:38 config-4.18.16-81.aws
lrwxrwxrwx  1 root root      29 Nov 30 01:23 default-aws -> org.clearlinux.aws.4.18.16-81
-rwxr-xr-x  1 root root 7557168 Oct 21 14:56 org.clearlinux.aws.4.18.16-80
-rwxr-xr-x  1 root root 7520304 Nov 30 01:23 org.clearlinux.aws.4.18.16-81

> syslinux.cfg is still pointing to older kernel
clear@clr-d43af41c51a5449ab6f5e76a00c3465e ~ $ cat /boot/syslinux.cfg 
DEFAULT org.clearlinux.aws.4.18.16-80
LABEL org.clearlinux.aws.4.18.16-80
  KERNEL org.clearlinux.aws.4.18.16-80
APPEND root=PARTUUID=3db33026-5289-4a39-b24b-6d4a59b03b7e console=tty0 console=ttyS0,115200n8 quiet init=/usr/bin/initra-aws initcall_debug tsc=reliable no_timer_check noreplace-smp rootfstype=ext4 cryptomgr.notests rcupdate.rcu_expedited=1 rw

bryteise, 2018-12-04:
Could you run CBM_DEBUG=1 clr-boot-manager update alone and paste the logs from that?

bktan8, 2018-12-04:
clear@clr-d43af41c51a5449ab6f5e76a00c3465e ~ $ CBM_DEBUG=1

clear@clr-d43af41c51a5449ab6f5e76a00c3465e ~ $ sudo clr-boot-manager update
[FATAL] cbm (../src/bootman/update.c:L393): Failed to install bootloader

bryteise, 2018-12-05:
@bktan8 Okay, so unfortunately this is back to a clearlinux distribution bug. The syslinux command `extlinux -U /boot` is failing for some reason and causing the kernel to not be updated.

bryteise, 2018-12-05:
At the same time clr-boot-manager should probably try and update kernels regardless of if the bootloader update works or not (though the update should be still marked as a failure). @ikeyd thoughts on that?

ikeyd, 2018-12-09:
@bryteise **perhaps** given that we'll always retain the currently running kernel. Perhaps we should make it explicitly clear that older kernels mightn't work properly. Additionally this should still be a fatal error in image mode. Last but not least we need to get rid of WARNING/ERROR/FATAL notation as it is very confusing. It either works or it doesn't. WARNING for all other cases.

bryteise, 2018-12-27:
@ikeyd Yes, I like the sound of all of that.

dorileo, 2019-09-18:
c-b-m v3.2 might fix this issue.