clearlinux/clr-boot-manager #134: Commit breaks LUKS on Solus

Girtablulu, 2018-12-08:
We had an issue update clr-boot-manger on Solus which broke installs which used LUKS after the system run clr-boot-manager update. 

I looked into it and found the commit which is causing this problem on solus

commit ef77136

after I patched this one out, I could update and use v3.1.0 and have no issues sofar on my LUKS test system. (Gonna keep an eye on it, @DataDrake is aware of it) 

Another small thing I found was running clr-boot-manager list-kernels on a UEFI system I get an error
```

[FATAL] cbm (../src/bootloaders/systemd-class.c:L507): sd_class_get_default_kernel Unable to parse loader config
  com.solus-project.current.4.19.7-103
```

Which does not happen on a GRUB install

bryteise, 2018-12-08:
Ah you might be hitting things that #129 resolves on listing kernels but that looks different so I'll have to try and bootup a Solus install and try that out. I'm really not sure what is going on with the LUKS breakage right now and I'll need to look into that because this does work for Clear Linux LUKS installs.

Girtablulu, 2018-12-10:
Ah okay, didn't see the PR sorry about that.

DataDrake didn't had time yet to look into it and figure out exactly why this happens, I just confirmed his suspicion that this commit is causing it.

cata0309, 2019-06-07:
The sudo clr-boot-manager list-kernels command still shows the error, any fix ?

bryteise, 2019-06-07:
@cata0309 Not yet sorry X(. Any chance you could paste a copy of the config that is showing the error?

cata0309, 2019-06-08:
It is a fresh install, not a personalized config

bryteise, 2019-06-10:
Hrm I seem to having the iso installer crash when running in a VM. Will need to poke at this more X(.

circlenaut, 2019-08-01:
Hey guys, might my [issue](https://github.com/clearlinux/clr-boot-manager/issues/166) be related to this?

bvdlingen, 2019-09-30:
Same issue here. 

Looks scary but everything works fine.

dorileo, 2020-01-31:
Please can you confirm this is still any issue. Can you please check with a more recent version (not sure the version Solus is shipping these days).

silkeh, 2020-07-29:
I can confirm this is still an issue. The cause seems to be that the glob in the linked commit has the wrong order:

```
$ ls -1 /sys/block/dm-2/slaves/*{,/slaves/*}/dev
/sys/block/dm-2/slaves/dm-0/dev
/sys/block/dm-2/slaves/dm-0/slaves/nvme0n1p2/dev
```

The glob should probably be: `/sys/block/dm-2/slaves/*{/slaves/*,}/dev`

PR #240 (2024-10-11): Fix LUKS glob order
As noted in #134, I believe the order is the wrong way round.

I have tested this on Solus with Btrfs on LUKS, and Btrfs on LVM on LUKS. I was unable to test this on a clean Clear Linux or Solus install, as both failed to install with disk encryption in a (BIOS) VM.

Resolves #134
closed 2024-10-11 by bryteise