clearlinux/clr-boot-manager #139: Error: Failed to parse given uname release: 4.19.13-1901141830.iot-lts2018-sos when updating ACRN setup

LibertadCruz, 2019-02-08:
LOG from the update:   [acrn_swupd_update.txt](https://github.com/clearlinux/swupd-client/files/2842695/acrn_swupd_update.txt)

**Description:** I am updating a previously configured ACRN environment on an apolo lake nuc. The swupd update does not fail, but it does show Errors involving the kernel being run. 

I opened this issue because I cannot determine if its an error or expected. But it does intrigue me since its the first time I am seing them. I have gone through the update procedure with ACRN and Clear linux since forever and this is the first time this errors are present.

Hardware specifications are described in the attached file ACRN_hardware_specifications.txt
[ACRN_hardware_specifications.txt](https://github.com/clearlinux/swupd-client/files/2842715/ACRN_hardware_specifications.txt)

The steps to setup a NUC with the ACRN can be found in this wiki:  https://projectacrn.github.io/latest/getting-started/apl-nuc.html

bryteise, 2019-02-08:
This is a content bug unfortunately as the release isn't valid. My memory is that this has been fixed in the kernel package.

LibertadCruz, 2019-02-11:
@otaviobp, so I have run the commands leading to the issue of #137, it gives me the same output. 
[kernel-iot-unparsed.txt](https://github.com/clearlinux/clr-boot-manager/files/2852609/kernel-iot-unparsed.txt)

So, I think is more of the same.

bryteise, 2019-02-11:
@LibertadCruz It looks like you are running a kernel with a bad release entry, clr-boot-manager tries to avoid breaking things it doesn't understand so it doesn't touch things with content it can't parse. You'll need to switch your booted kernel into a good version and then manually clean up the invalid kernels.

gvancuts, 2019-02-14:
I was wrong. I now have a system that does **not** have any of those older kernels with the broken release field. But I am getting the same error as reported by @LibertadCruz. I actually bumped into this when installing a new kernel bundle. I only see this when running on top of ACRN (so running the `iot-lts2018-sos` kernel). If I reboot the exact same system and run Clear Linux natively, I do not get any error.

I should note, in case you do not know those details, that booting under ACRN is slightly different and while `systemd-boot` is used, a special entry (`acrn.conf`) is added manually - see https://projectacrn.github.io/latest/getting-started/apl-nuc.html#add-the-acrn-hypervisor-to-the-efi-partition for more details. I do not know whether this matters or not.

Here is the output in both scenarii:

### Running on top of ACRN
```
$ sudo swupd bundle-add kernel-native
Downloading packs...
	...100%
Starting download of remaining update content. This may take a while...
	...100%
Finishing download of update content...
Installing bundle(s) files...
	...100%
Calling post-update helper scripts.
none
[ERROR] cbm (../src/bootman/bootman.c:L681): Failed to parse given uname release: 4.19.19-9.iot-lts2018-sos
[WARNING] cbm (../src/bootman/bootman.c:L70): Unable to parse the currently running kernel: 4.19.19-9.iot-lts2018-sos
[ERROR] cbm (../src/bootman/update.c:L196): Cannot determine the currently running kernel
Successfully installed 1 bundle
gvancuts@cs-corei5 ~ $ uname -r
4.19.19-9.iot-lts2018-sos
```

### Running native Clear Linux
```
$ sudo swupd bundle-add kernel-native
Downloading packs...
	...100%
Starting download of remaining update content. This may take a while...
	...100%
Finishing download of update content...
Installing bundle(s) files...
	...100%
Calling post-update helper scripts.
none
Successfully installed 1 bundle
gvancuts@cs-corei5 ~ $ uname -r
4.19.20-420.lts
```

I'm running clr-boot-manager version 3.1.0:
```
$ clr-boot-manager version
clr-boot-manager - version 3.1.0

Copyright © 2016-2017 Intel Corporation

clr-boot-manager is free software; you can redistribute it and/or modify
it under the terms of the GNU Lesser General Public License as published by
the Free Software Foundation; either version 2.1 of the License, or
(at your option) any later version.
```

busykai, 2019-02-15:
@bryteise [this function](https://github.com/clearlinux/clr-boot-manager/blob/master/src/bootman/kernel.c#L390) is the culprit: it [limits kernel type to effectively 14](https://github.com/clearlinux/clr-boot-manager/blob/d4d2cbf7f8f4e3d555a085c39f837ee50483f71d/src/bootman/kernel.c#L452) and 'iot-lts2018-sos' (15 chars) does not fit. There are two issues: 1) the check is incorrect since it allocates 16 bytes (`CBM_KELEM_LEN + 1`), but checks if length of element is `>= CBM_KELEM_LEN` (15) instead of `>`; 2) allowed size of the kernel type is not coordinated with our packaging/naming.

bryteise, 2019-02-20:
@busykai Ah thanks!

PR #141 (2019-02-20): Fix kernel type handling/size.
Fixes #139.

- Fix size check.
- Increase max len from 15 to 31.

@bryteise
closed 2019-02-20 by bryteise