clearlinux/clr-boot-manager #140: Failed to parse uname release for Clear on WSL

jwakre, 2019-02-15:
When running Clear on WSL (Windows Subsystem for Linux), Swupd updates fail when calling the clr-boot-manager. It looks like cbm_parse_system_kernel is confused by the kernel naming scheme. More information and error snippets can be found at clearlinux/swupd-client#791

Snippet from log for update from 24920->25760:
.
.
.
Applying update
        ...100%
Update was applied.
Calling post-update helper scripts.
none
[ERROR] cbm (../src/bootman/bootman.c:L447): Failed to parse given uname release: 4.4.0-43-Microsoft
[WARNING] cbm (../src/bootman/bootman.c:L68): Unable to parse the currently running kernel: 4.4.0-43-Microsoft
[ERROR] cbm (../src/lib/files.c:L165): Invalid block device: /
[ERROR] cbm (../src/lib/system_stub.c:L31): Invalid block device: 0:2
Out of memory[ERROR] cbm (../src/bootman/update.c:L255): No kernels discovered in ///usr/lib/kernel, bailing
Update took 1369.5 seconds
3262 files were not in a pack
Update successful. System updated from version 24920 to version 25760
.
.
.

DaanDeMeyer, 2020-08-01:
Similar problem happening when trying to build clear linux images on Github Actions:

```
[ERROR] cbm (../src/bootman/bootman.c:L740): Failed to parse given uname release: 5.4.0-1020-azure
[WARNING] cbm (../src/bootman/bootman.c:L90): Unable to parse the currently running kernel: 5.4.0-1020-azure
```

Fubuchi, 2022-05-07:
Since I thought clr-boot-manager have no use in WSL environment, I created a patch version that by skip all the checks that generated the errors: https://github.com/Fubuchi/clr-boot-manager

The pre-compiled version is for Solus WSL, so you may need to compile it your selft if you want to use it.