clearlinux/clr-boot-manager #142: Failed to install bootloader on GCE

FrankFFFF, 2019-02-20:
I'm trying to add something to cmdline by this method: https://clearlinux.org/documentation/clear-linux/guides/maintenance/kernel-modules

And when i run `sudo clr-boot-manager update`, it gives me this `[FATAL] cbm (../src/bootman/update.c:L396): Failed to install bootloader`.

I'm running clear-27910-gce on GCE, does it require something else?

bryteise, 2019-02-20:
Ah we don't usually test updates in that environment and apparently the syslinux installation method being used is not being very happy in GCE. I'll take a look at this.

zardoz-bzar, 2019-06-23:
https://github.com/clearlinux/clr-boot-manager/issues/147  
  
Same bug basically?  I've been trying to get Clear production ready on GCE. Have a pull request for another nit. Trying to figure out the rest of the papercuts.

zardoz-bzar, 2019-07-04:
I recommend closing this is as a non-bug since functionality is in no way impaired for the runtime environment, syslinux.cfg is updated properly, and GCE works fine.

It's an aesthetic issue at best and and bug #147 supercedes this.

bryteise, 2019-09-03:
Okay, well we are working on improving syslinux support but the ux issue is probably a longer term thing. I'll close this as it isn't causing breakage now.

closed 2019-09-03 by bryteise