clearlinux/clr-boot-manager #147: clr-boot-manager update segfaults when /boot is not mounted on systems without UEFI

mpurzynski, 2019-03-25:
On a system without UEFI, with standard BIOS, GPT and syslinux, the /boot partition is not automatically mounted during the clr-boot-manager update

This uncovered some bug in the clr-boot-manager code. If /boot/syslinux.cfg is not present, the clr-boot-manager segfaults

`
[337648.001907] clr-boot-manage[15656]: segfault at ffffffffffffffc0 ip 00007fddd239c0e4 sp 00007ffd21df6bf8 error 5 in libc-2.29.so[7fddd2310000+174000]
[337648.001915] Code: c1 e1 20 48 09 ca 48 89 f9 48 31 c1 48 83 e0 c0 48 d3 fa 48 85 d2 74 7e 48 0f bc c2 c3 66 0f 1f 84 00 00 00 00 00 48 83 e0 c0 <66> 0f 74 00 66 0f 74 48 10 66 0f 74 50 20 66 0f 74 58 30 66 0f d7
`

`
root@nsm2 /home/michal # clr-boot-manager update
[FATAL] cbm (../src/bootman/update.c:L396): Failed to install bootloader
Segmentation fault (core dumped)
`

A quick debug session hints at a possibly uninitialized string, if the /boot/syslinux.cfg cannot be found

`
(gdb) run update
Starting program: /usr/bin/clr-boot-manager update
[Detaching after vfork from child process 22232]
[FATAL] cbm (../src/bootman/update.c:L396): Failed to install bootloader

Program received signal SIGSEGV, Segmentation fault.
__strlen_sse2 () at ../sysdeps/x86_64/multiarch/../strlen.S:152
152 ../sysdeps/x86_64/multiarch/../strlen.S: No such file or directory.
(gdb) where
#0  __strlen_sse2 () at ../sysdeps/x86_64/multiarch/../strlen.S:152
#1  0x00007ffff7df8e32 in __GI___strdup (s=0xffffffffffffffff <error: Cannot access memory at address 0xffffffffffffffff>)
    at strdup.c:41
#2  0x000000000040bf09 in file_get_text (path=path@entry=0x4307d0 "/boot/syslinux.cfg", out_buf=out_buf@entry=0x7fffffffc300)
    at ../src/lib/files.c:301
#3  0x000000000041128e in syslinux_set_default_kernel (manager=0x41fe10, default_kernel=0x52e6d0)
    at ../src/bootloaders/syslinux.c:193
#4  0x0000000000406775 in boot_manager_set_default_kernel (self=self@entry=0x41fe10, kernel=kernel@entry=0x52e6d0)
    at ../src/bootman/bootman.c:336
#5  0x000000000040a07f in boot_manager_update_native (self=0x41fe10) at ../src/bootman/update.c:339
#6  boot_manager_update (self=self@entry=0x41fe10) at ../src/bootman/update.c:54
#7  0x00000000004058a8 in cbm_command_update (argc=<optimized out>, argv=<optimized out>) at ../src/cli/ops/update.c:77
#8  0x000000000040475b in main (argc=0, argv=0x7fffffffc658) at ../src/cli/main.c:276

(gdb) list
147 in ../sysdeps/x86_64/multiarch/../strlen.S
`

Manually mounting /dev/sda1 as /boot helps

(gdb) run update
Starting program: /usr/bin/clr-boot-manager update
[Detaching after vfork from child process 22414]
[Inferior 1 (process 22305) exited normally]

bryteise, 2019-03-25:
Unfortunately we don't really support non-MBR, syslinux where the rootfs doesn't also contain boot. I'd suggest keeping the boot partition mounted and hopefully it works because I don't forsee us adding support any time soon to detect the boot partition in a non-UEFI setup.

mpurzynski, 2019-05-13:
Nowadays we get a segmentation fault even with /boot mounted

bryteise, 2019-05-13:
Ugh okay, well I'll take a look and see if I can figure this out. Thanks for the update.

bryteise, 2019-09-03:
Should be fixed in latest release.

closed 2019-09-03 by bryteise