clearlinux/clr-boot-manager #148: clr-boot-manager coredump if /dev/sdb1 was mounted beforehand

LizYou, 2019-04-12:
intel@fr /usr/lib/kernel $ sudo CBM_DEBUG=1 clr-boot-manager update 
[INFO] cbm (../src/bootman/bootman.c:L685): Current running kernel: 5.0.7-729.native
[INFO] cbm (../src/bootman/sysconfig.c:L98): Discovered UEFI ESP: /dev/disk/by-partuuid/e7c904fe-abba-4dd8-913e-c93a1d18331f
[INFO] cbm (../src/bootman/sysconfig.c:L123): Fully resolved boot device: /dev/sdb1 
[DEBUG] cbm (../src/bootman/bootman.c:L132): UEFI boot now selected (shim-systemd) 
[INFO] cbm (../src/bootman/bootman.c:L705): path ///usr/lib/initrd.d does not exist 
[INFO] cbm (../src/bootman/bootman.c:L416): Checking for mounted boot dir 
[DEBUG] cbm (../src/bootman/bootman.c:L437): Boot device already mounted at /mnt
[INFO] cbm (../src/bootman/bootman.c:L444): Skipping to native update 
[DEBUG] cbm (../src/bootman/update.c:L171): Now beginning update_native 
[DEBUG] cbm (../src/bootman/update.c:L180): update_native: 1 available kernels
[DEBUG] cbm (../src/bootman/update.c:L200): update_native: Running kernel is (native) ///usr/lib/kernel/org.clearlinux.native.5.0.7-729
[SUCCESS] cbm (../src/bootman/update.c:L212): update_native: Bootloader updated [SUCCESS] cbm (../src/bootman/update.c:L228): update_native: Repaired running kernel ///usr/lib/kernel/org.clearlinux.native.5.0.7-729
[DEBUG] cbm (../src/bootman/update.c:L237): update_native: Checking kernels for type native [INFO] cbm (../src/bootman/update.c:L252): update_native: Default kernel for type native is ///usr/lib/kernel/org.clearlinux.native.5.0.7-729
[SUCCESS] cbm (../src/bootman/update.c:L264): update_native: Installed tip for native: ///usr/lib/kernel/org.clearlinux.native.5.0.7-729
[SUCCESS] cbm (../src/bootman/update.c:L278): update_native: Installed last_good kernel (native) (///usr/lib/kernel/org.clearlinux.native.5.0.7-729)
[DEBUG] cbm (../src/bootman/update.c:L289): update_native: Analyzing for type native: ///usr/lib/kernel/org.clearlinux.native.5.0.7-729
[DEBUG] cbm (../src/bootman/update.c:L292): update_native: Skipping running kernel
[INFO] cbm (../src/bootman/bootman.c:L416): Checking for mounted boot dir 
[INFO] cbm (../src/bootman/bootman.c:L419): boot_dir is already mounted: /mnt 
[SUCCESS] cbm (../src/bootman/update.c:L347): update_native: Default kernel for native is ///usr/lib/kernel/org.clearlinux.native.5.0.7-729
[DEBUG] cbm (../src/bootman/update.c:L360): No kernel removals found
free(): double free detected in tcache 2 Aborted

LizYou, 2019-04-12:
After I umount the /mnt, the problem was gone.

mrkz, 2019-07-17:
@bryteise pending close with #153 merged (?)

bryteise, 2019-09-03:
Yes, should be fixed now.

closed 2019-09-03 by bryteise