clearlinux/clr-boot-manager #150: `/etc/kernel/cmdline-removal.d` does not behave as expected

chipb, 2019-05-09:
**Describe the bug**
I'm trying to remove a kernel command line argument (`console=ttyS0,115200n8`) from my boot entries, but the method provided by clr-boot-manager doesn't appear to work as expected.

I've added a file to `/etc/kernel/cmdline-removal.d`:
```sh
$ cat /etc/kernel/cmdline-removal.d/ttyS0.conf
console=ttyS0,115200n8
```
...and refreshed the entries, but the argument persists:
```sh
$ sudo clr-boot-manager update
$ sudo sh -c 'grep -o "console=ttyS0[^ ]*" /boot/loader/entries/*.conf'
/boot/loader/entries/Clear-linux-native-5.0.6-726.conf:console=ttyS0,115200n8
/boot/loader/entries/Clear-linux-native-5.0.7-727.conf:console=ttyS0,115200n8
```
Rudimentary strace debug shows the `ttyS0.conf` file is opened and read. For what it's worth (probably not much), it also shows that the reading of `/etc/kernel/cmdline-removal.d` occurs well before `/usr/lib/kernel/cmdline-5.0.7-727.native` is read, so my initial instinct is that perhaps the feature is only meant to drop arguments not provided by the kernel bundle's "innate" cmdline?

**Expected behavior**
As documented in `clr-boot-manager(1)`, it sounds like my `/etc/kernel/cmdline-removal.d/ttyS0.conf` should remove the unwanted `console=` argument.

**Environment (please complete the following information):**
 - Clear Linux OS version: [`cat /usr/lib/os-release`]
```
NAME="Clear Linux OS"
VERSION=1
ID=clear-linux-os
ID_LIKE=clear-linux-os
VERSION_ID=28710
PRETTY_NAME="Clear Linux OS"
ANSI_COLOR="1;35"
HOME_URL="https://clearlinux.org"
SUPPORT_URL="https://clearlinux.org"
BUG_REPORT_URL="mailto:dev@lists.clearlinux.org"
PRIVACY_POLICY_URL="http://www.intel.com/privacy"
```
 - Bundles: [e.g. `sudo swupd bundle-list`]
```
$ sudo swupd bundle-list | grep -v dev
Babel
NetworkManager
Remmina
Sphinx
acpica-unix2
akonadi
alsa-utils
ansible
ark
atom
baobab
bc
binutils
bison
boot-encrypted
bootloader
bootloader-extras
c-basic
cabextract
ccache
ceph
cheese
cloc
clr-network-troubleshooter
containers-basic
containers-virt
cpio
cryptography
cryptoprocessor-management
curl
darktable
desktop
desktop-apps
desktop-apps-extras
desktop-assets
desktop-autostart
desktop-gnomelibs
desktop-kde-apps
desktop-kde-games
desktop-kde-libs
desktop-locales
diffoscope
diffutils
digikam
dnf
docbook-utils
docutils
dolphin
dosfstools
doxygen
editors
emacs
enterprise-login
eog
ethtool
evince
evolution
file
file-roller
findutils
firefox
firmware-update
flatpak
flex
fontforge
fonts-basic
fonttools
fuse
fwupdate
gdb
geany
gedit
gimp
git
gjs
glibc-locale
global
gnome-base-libs
gnome-calculator
gnome-characters
gnome-color-manager
gnome-disk-utility
gnome-font-viewer
gnome-logs
gnome-music
gnome-photos
gnome-screenshot
gnome-system-monitor
gnome-todo
gnome-weather
go-basic
gpgme
gphoto2
graphviz
gstreamer
gtk-vnc
gvim
gwenview
gzip
hardware-printing
hardware-uefi
hexchat
htop
httpd
icdiff
inotify-tools
intltool
ipmitool
iproute2
iptables
irssi
joe
kamera
kate
kbd
kcalc
kde-frameworks5
kernel-install
kernel-native
kleopatra
koji
konqueror
konsole
konversation
krita
kvm-host
less
lib-imageformat
lib-opengl
lib-openssl
lib-samba
libX11client
libarchive
libglib
libstdcpp
libva-utils
libxml2
libxslt
linux-firmware
linux-firmware-extras
linux-firmware-wifi
linux-tools
llvm
mail-utils
make
man-pages
minicom
mixer
mkosi
mutt
nasm
nautilus
network-basic
nfs-utils
nginx
nodejs-basic
notmuch
offlineimap
okular
openblas
openldap
openssh-server
openssl
os-clr-on-clr
os-core
os-core-update
os-core-webproxy
os-installer
p11-kit
package-builder
package-utils
parallel
parted
patch
perl-basic
pidgin
plzip
pmdk
polkit
postgresql
powertop
pulseaudio
pygobject
python2-basic
python3-basic
qemu-guest-additions
qt-basic
qt-core
quassel
restic
ruby-basic
samba
scons
seahorse
shells
smartmontools
spectacle
spice-gtk
storage-utils
strace
subversion
sudo
syndication
sysadmin-basic
telemetrics
texinfo
texlive
thermal_daemon
thunderbird
tigervnc
tmux
totem
tzdata
unbundle
unzip
user-basic
valgrind
vim
vinagre
virt-manager
virt-manager-gui
virt-viewer
vlc
web-server-basic
webkitgtk
weechat
wget
which
wpa_supplicant
x11-server
x11-tools
x11vnc
xemacs
xfsprogs
xterm
xz
yakuake
yasm
zip
znc
zsh
zstd
```

bryteise, 2019-05-10:
Fixed in #151

bryteise, 2019-05-13:
Note for Clear Linux this is in release 29360

closed 2019-05-13 by bryteise

chipb, 2019-05-13:
Hm. Pretty sure it was post-29360, because I unsuccessfully checked with it first, but it's definitely working after my update to 29390. :+1: 

Thanks again, William.

bryteise, 2019-05-14:
Gah I probably just mixed up the release numbers, sorry. Glad it is working now =).