clearlinux/clr-boot-manager #152: Program exiting with SIGABRT error: "double free or corruption (!prev)"

chax, 2019-05-22:
Distro: Solus
Built from latest commit bd30daa

Running cbm update i get this:
```
==22972== Memcheck, a memory error detector
==22972== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==22972== Using Valgrind-3.13.0 and LibVEX; rerun with -h for copyright info
==22972== Command: clr-boot-manager update
==22972== 
[INFO] cbm (../src/bootman/bootman.c:L685): Current running kernel: 5.0.16-116.current
[INFO] cbm (../src/bootman/sysconfig.c:L98): Discovered UEFI ESP: /dev/disk/by-partuuid/64edc316-0c41-4cda-bac4-6d9b0b8e3395
[INFO] cbm (../src/bootman/sysconfig.c:L123): Fully resolved boot device: /dev/sda1
[DEBUG] cbm (../src/bootman/bootman.c:L132): UEFI boot now selected (systemd)
[INFO] cbm (../src/bootman/bootman.c:L705): path ///usr/lib/initrd.d does not exist
[INFO] cbm (../src/bootman/bootman.c:L416): Checking for mounted boot dir
[DEBUG] cbm (../src/bootman/bootman.c:L437): Boot device already mounted at /boot/efi
[INFO] cbm (../src/bootman/bootman.c:L444): Skipping to native update
[DEBUG] cbm (../src/bootman/update.c:L171): Now beginning update_native
[DEBUG] cbm (../src/lib/cmdline.c:L380): Removing cmdline using file: ///etc/kernel/cmdline-removal.d/99-silent.conf
[DEBUG] cbm (../src/lib/cmdline.c:L380): Removing cmdline using file: ///etc/kernel/cmdline-removal.d/99-silent.conf
[DEBUG] cbm (../src/lib/cmdline.c:L380): Removing cmdline using file: ///etc/kernel/cmdline-removal.d/99-silent.conf
[DEBUG] cbm (../src/bootman/update.c:L180): update_native: 3 available kernels
[DEBUG] cbm (../src/bootman/update.c:L200): update_native: Running kernel is (current) ///usr/lib/kernel/com.solus-project.current.5.0.16-116
[SUCCESS] cbm (../src/bootman/update.c:L212): update_native: Bootloader updated
[SUCCESS] cbm (../src/bootman/update.c:L228): update_native: Repaired running kernel ///usr/lib/kernel/com.solus-project.current.5.0.16-116
[DEBUG] cbm (../src/bootman/update.c:L237): update_native: Checking kernels for type lts
[INFO] cbm (../src/bootman/update.c:L252): update_native: Default kernel for type lts is ///usr/lib/kernel/com.solus-project.lts.4.9.176-131
[SUCCESS] cbm (../src/bootman/update.c:L264): update_native: Installed tip for lts: ///usr/lib/kernel/com.solus-project.lts.4.9.176-131
[SUCCESS] cbm (../src/bootman/update.c:L278): update_native: Installed last_good kernel (lts) (///usr/lib/kernel/com.solus-project.lts.4.9.141-116)
[DEBUG] cbm (../src/bootman/update.c:L289): update_native: Analyzing for type lts: ///usr/lib/kernel/com.solus-project.lts.4.9.176-131
[INFO] cbm (../src/bootman/update.c:L295): update_native: not-running: ///usr/lib/kernel/com.solus-project.lts.4.9.176-131
[DEBUG] cbm (../src/bootman/update.c:L300): update_native: Skipping default-lts: ///usr/lib/kernel/com.solus-project.lts.4.9.176-131
[DEBUG] cbm (../src/bootman/update.c:L289): update_native: Analyzing for type lts: ///usr/lib/kernel/com.solus-project.lts.4.9.141-116
[INFO] cbm (../src/bootman/update.c:L295): update_native: not-running: ///usr/lib/kernel/com.solus-project.lts.4.9.141-116
[INFO] cbm (../src/bootman/update.c:L305): update_native: not-default-lts: ///usr/lib/kernel/com.solus-project.lts.4.9.141-116
[DEBUG] cbm (../src/bootman/update.c:L308): update_native: Skipping last_good kernel
[DEBUG] cbm (../src/bootman/update.c:L237): update_native: Checking kernels for type current
[INFO] cbm (../src/bootman/update.c:L252): update_native: Default kernel for type current is ///usr/lib/kernel/com.solus-project.current.5.0.16-116
[SUCCESS] cbm (../src/bootman/update.c:L264): update_native: Installed tip for current: ///usr/lib/kernel/com.solus-project.current.5.0.16-116
[SUCCESS] cbm (../src/bootman/update.c:L278): update_native: Installed last_good kernel (current) (///usr/lib/kernel/com.solus-project.current.5.0.16-116)
[DEBUG] cbm (../src/bootman/update.c:L289): update_native: Analyzing for type current: ///usr/lib/kernel/com.solus-project.current.5.0.16-116
[DEBUG] cbm (../src/bootman/update.c:L292): update_native: Skipping running kernel
[DEBUG] cbm (../src/lib/cmdline.c:L380): Removing cmdline using file: ///etc/kernel/cmdline-removal.d/99-silent.conf
[DEBUG] cbm (../src/lib/cmdline.c:L380): Removing cmdline using file: ///etc/kernel/cmdline-removal.d/99-silent.conf
[DEBUG] cbm (../src/lib/cmdline.c:L380): Removing cmdline using file: ///etc/kernel/cmdline-removal.d/99-silent.conf
[INFO] cbm (../src/bootman/bootman.c:L416): Checking for mounted boot dir
[INFO] cbm (../src/bootman/bootman.c:L419): boot_dir is already mounted: /boot/efi
[SUCCESS] cbm (../src/bootman/update.c:L347): update_native: Default kernel for current is ///usr/lib/kernel/com.solus-project.current.5.0.16-116
[DEBUG] cbm (../src/bootman/update.c:L360): No kernel removals found
==22972== Invalid free() / delete / delete[] / realloc()
==22972==    at 0x483897B: free (in /usr/lib64/valgrind/vgpreload_memcheck-amd64-linux.so)
==22972==    by 0x407C84: _autofree_func_char (update.c:55)
==22972==    by 0x407C84: boot_manager_update (update.c:35)
==22972==    by 0x403B18: cbm_command_update (update.c:77)
==22972==    by 0x4029A4: main (main.c:276)
==22972==  Address 0x4d9f010 is 0 bytes inside a block of size 4,096 free'd
==22972==    at 0x483897B: free (in /usr/lib64/valgrind/vgpreload_memcheck-amd64-linux.so)
==22972==    by 0x404512: _autofree_func_char (util.h:55)
==22972==    by 0x404512: mount_boot (bootman.c:400)
==22972==    by 0x407BB2: boot_manager_update (update.c:50)
==22972==    by 0x403B18: cbm_command_update (update.c:77)
==22972==    by 0x4029A4: main (main.c:276)
==22972==  Block was alloc'd at
==22972==    at 0x483774F: malloc (in /usr/lib64/valgrind/vgpreload_memcheck-amd64-linux.so)
==22972==    by 0x4917B1E: realpath@@GLIBC_2.3 (canonicalize.c:78)
==22972==    by 0x4042F4: realpath (stdlib.h:48)
==22972==    by 0x4042F4: boot_manager_get_boot_dir (bootman.c:557)
==22972==    by 0x404457: mount_boot (bootman.c:409)
==22972==    by 0x407BB2: boot_manager_update (update.c:50)
[... 44 lines cut ...]
==23031==  Block was alloc'd at
==23031==    at 0x483774F: malloc (in /usr/lib64/valgrind/vgpreload_memcheck-amd64-linux.so)
==23031==    by 0x4917B1E: realpath@@GLIBC_2.3 (canonicalize.c:78)
==23031==    by 0x4042F4: realpath (stdlib.h:48)
==23031==    by 0x4042F4: boot_manager_get_boot_dir (bootman.c:557)
==23031==    by 0x404457: mount_boot (bootman.c:409)
==23031==    by 0x404BCF: boot_manager_list_kernels (bootman.c:515)
==23031==    by 0x403064: cbm_command_list_kernels (kernels.c:70)
==23031==    by 0x4029A4: main (main.c:276)
==23031== 
  com.solus-project.lts.4.9.176-131
  com.solus-project.lts.4.9.141-116
* com.solus-project.current.5.0.16-116
==23031== 
==23031== HEAP SUMMARY:
==23031==     in use at exit: 0 bytes in 0 blocks
==23031==   total heap usage: 826 allocs, 827 frees, 4,112,945 bytes allocated
==23031== 
==23031== All heap blocks were freed -- no leaks are possible
==23031== 
==23031== For counts of detected and suppressed errors, rerun with: -v
==23031== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
```

I hope this is enough info for you to fix this. If you need any additional info, just let me know.

bryteise, 2019-05-22:
That should be plenty of information thanks for the detailed report!

chax, 2019-05-23:
I think this might be duplicate of #148. I tried to unmount my boot partition like suggested in that issue, and no more crashing after my efi partition is unmounted.

chax, 2019-05-23:
Found it.
i can do PR if you want.

chax, 2019-05-23:
here's a patch:

``` diff
diff --git a/src/bootman/bootman.c b/src/bootman/bootman.c
index e715c22..fc785e0 100644
--- a/src/bootman/bootman.c
+++ b/src/bootman/bootman.c
@@ -442,7 +442,7 @@ int mount_boot(BootManager *self, char **boot_directory)
                 }
                 /* Successfully using their premounted ESP, go use it */
                 LOG_INFO("Skipping to native update");
-                *boot_directory = boot_dir;
+                *boot_directory = strdup(boot_dir);
                 if (*boot_directory) {
                         ret = 0;
                 }
```

chax, 2019-06-05:
Fixed with #153

closed 2019-06-05 by chax