clearlinux/clr-boot-manager #154: Invalid block device: / : Out of memory: No kernels discovered

devimc, 2019-06-04:
**Describe the bug**
```
[ERROR] cbm (../src/lib/files.c:L165): Invalid block device: /
[ERROR] cbm (../src/lib/system_stub.c:L31): Invalid block device: 0:54
Out of memory[ERROR] cbm (../src/bootman/update.c:L176): No kernels discovered in ///usr/lib/kernel, bailing
Successfully installed 1 bundle
```

**To Reproduce**
Run `swupd bundle-add sysadmin-basic` inside a docker container

**Expected behavior**
No errors

**Environment (please complete the following information):**
 - Clear Linux OS Version: 29280
 - Platform: docker

miguelinux, 2019-06-04:
> **Environment (please complete the following information):**
>  - Clear Linux OS Version: 29280
>  - Platform: docker

This issue should be solved at 29530.

Can you test in that version please.

devimc, 2019-06-04:
@miguelinux the latest container version of clear is 29280, I think the maintainer should update it

bryteise, 2019-06-04:
Given this is fixed in cbm I'm closing the issue. The bug is visual as the container case doesn't even want cbm to run so I don't think we'll likely be getting a new docker image for this issue alone.

closed 2019-06-04 by bryteise