clearlinux/clr-boot-manager #158: clr-boot-manager does a double free if /dev/sda1 (EFI System) partition is already mounted

bktan8, 2019-07-16:
If /dev/sda1 (EFI System) is mounted and you call "sudo clr-boot-manager list-kernels", it outputs a nasty message like this:

``sudo mount /dev/sda1 /mnt``

``user@clr ~ $ sudo clr-boot-manager list-kernels``

``double free or corruption (!prev)
Aborted``

Would be nice to get a more graceful message and offer a suggestion the workaround.

ahkok, 2019-07-16:
@bktan8
- what kernels do you have installed?
- what is the output when you don't have `/mnt` mounted?

bktan8, 2019-07-16:
@ahkok 

* what kernels do you have installed?
``
bktan1@clr-0028be8d4d114d0b8f3a5b2ea1e3b701~ $ sudo mount /dev/sda1 /mnt
Password: 

bktan1@clr-0028be8d4d114d0b8f3a5b2ea1e3b701~ $ ls /mnt
EFI  loader

bktan1@clr-0028be8d4d114d0b8f3a5b2ea1e3b701~ $ sudo clr-boot-manager list-kernels
double free or corruption (!prev)
Aborted

bktan1@clr-0028be8d4d114d0b8f3a5b2ea1e3b701~ $ uname -a
Linux clr-0028be8d4d114d0b8f3a5b2ea1e3b701 5.1.17-796.native #1 SMP Thu Jul 11 05:52:46 PDT 2019 x86_64 GNU/Linux
``

* what is the output when you don't have /mnt mounted?
``
bktan1@clr-0028be8d4d114d0b8f3a5b2ea1e3b701~ $ sudo umount /mnt

bktan1@clr-0028be8d4d114d0b8f3a5b2ea1e3b701~ $ sudo clr-boot-manager list-kernels
  org.clearlinux.native.5.1.16-796
* org.clearlinux.native.5.1.17-796
``

bryteise, 2019-07-17:
Ah interesting, you have the EFI partition mounted outside of /boot and that seems to be the root cause of the issue. Thanks for the report.

bryteise, 2019-07-17:
Well I suck and didn't take a fix in master into Clear Linux. I blame being out and having no sleep X(. Likely should be fixed in a Thursday release.

closed 2019-07-17 by bryteise