clearlinux/clr-boot-manager #166: Changes to entries config don't persist after update

circlenaut, 2019-08-01:
After mounting my ESP partition and editing ESP/loader/entries/Solus-*.conf to change rd.luks.uuid=yyy to rd.luks.name=yyy=SolusCrypto and running clr-boot-manager update the entries revert back to rd.luks.uuid.

Running clor-boot-manager list-kernels also throws the following:
`[FATAL] cbm (../src/bootloaders/systemd-class.c:L507): sd_class_get_default_kernel Unable to parse loader config
  com.solus-project.lts.4.9.186-134
  com.solus-project.current.5.2.2-122
`

Might this be related to [this](https://github.com/clearlinux/clr-boot-manager/issues/134)?

I'm running version 3.1.0. Has the commit posted by Girtablulu been pushed? Is there a newer version out?

miguelinux, 2019-08-01:
Well currently this is how clr-boot-manger is programed, it will write the entry *.conf whit the UUID from the encrypted partition, and it wont add any new label.

That is a new enhancement for clr-boot-manger, to label the encrypted partition with a configuration file, but

What is the **use case** to label the UUID?

dorileo, 2020-01-31:
I suppose adding your kernel args to `/etc/kernel/cmdline` should be enough. Please let me know if it doesn't fix or issue (I'm not sure if the version Solus is running would have the feature though). Thanks.