clearlinux/clr-boot-manager #177: CBM should support labeling an EFI boot entry so it's easier to identify a CL partition

bktan8, 2019-09-19:
Right now, if there are multiple partitions on a hard drive, it's hard to know which one is Clear Linux or some other OS.  It would be nice if CBM can label a Clear partition with the version number as well.

dorileo, 2019-09-23:
What's the point of having the version number in the label?

ahkok, 2019-09-23:
The version number will continuously change, and therefore be wrong most of the time.

No issue with the installer labelling the rootfs during installation, but this should not be a `clr-boot-manager` thing.

ahkok, 2019-09-23:
Ah you mean `label` the menu entry. Sorry, misread.

ahkok, 2019-09-23:
I'd think that the boot menu labels already properly distinguish between `Clear Linux OS` and other OSs....

mdhorn, 2019-09-23:
I believe @bktan8 complaint has to do with this image:
[image]

Note that cbm added three entries -- all show precisely the same even though they are booting differently.  Also, this does not truly identify the OS as Clear Linux* OS.

dorileo, 2020-01-31:
In the commit c4e4903b4623682b0c2914a7eb54aef9ef6b63c9 I introduced a build flag to do that, we've not changed the packaging yet to pass the flag with something like "Clear Linux OS". I assume this would be enough.