clearlinux/clr-boot-manager #179: bootload manager errors

adriank41, 2019-10-01:
I have this errors when I run :

sudo clt-boot-manager update:
[ERROR] cbm (…/src/lib/bootvar.c:L319): blkid_probe_get_partitions() error
[ERROR] cbm (…/src/lib/bootvar.c:L319): blkid_probe_get_partitions() error
[ERROR] cbm (…/src/bootloaders/shim-systemd.c:L284): Cannot create EFI variable (boot entry)
[ERROR] cbm (…/src/bootloaders/shim-systemd.c:L285): Please manually update your bios to add a boot entry for Clear Linux

I run the 5.2.15-834.native kernel and cannot update to the latest kernel

sudo ctr-boot-manager list-kernels:

    *org.clearlinux.native.5.3.1-843
    org.clearlinux.native.5.2.15-834

dorileo, 2019-10-02:
hi @adriank41 can you run `blkid` and send me the output?

adriank41, 2019-10-02:
hi  dorileo,here is: 

dev/sda1: LABEL_FATBOOT="boot" LABEL="boot" UUID="CD0C-2361" TYPE="vfat" PARTLABEL="EFI" PARTUUID="d2633b4c-395e-419b-b123-56c0029a5288"
/dev/sda2: LABEL="swap" UUID="a70e7bb1-b139-47c7-b6e2-93a1017e795f" TYPE="swap" PARTLABEL="linux-swap" PARTUUID="8ea7f69e-fb1c-4aee-b529-211ca9da5c9b"
/dev/sda3: UUID="81039b8f-c249-4dea-a2ef-97381cc1bf2b" TYPE="crypto_LUKS" PARTLABEL="/" PARTUUID="13cec2dd-c2a4-4211-85fd-ca1fd459b289"
/dev/mapper/luks-81039b8f-c249-4dea-a2ef-97381cc1bf2b: LABEL="root" UUID="67a6acd9-8a33-4f51-a0d3-88c4ea3218ac" TYPE="ext4"

dorileo, 2019-10-02:
@adriank41 great, can you run `blkid -p /dev/sda` and send me the output ?

adriank41, 2019-10-02:
/dev/sda: PTUUID="62f0a216-f04c-41f8-a48c-5248c98281c5" PTTYPE="gpt"

dorileo, 2019-10-02:
@adriank41 sorry we should have ran it against the esp partition, can you run it again as: `blkid -p /dev/sda1`.

adriank41, 2019-10-02:
dev/sda1: LABEL_FATBOOT="boot" LABEL="boot" UUID="CD0C-2361" VERSION="FAT32" TYPE="vfat" USAGE="filesystem" PART_ENTRY_SCHEME="gpt" PART_ENTRY_NAME="EFI" PART_ENTRY_UUID="d2633b4c-395e-419b-b123-56c0029a5288" PART_ENTRY_TYPE="c12a7328-f81f-11d2-ba4b-00a0c93ec93b" PART_ENTRY_NUMBER="1" PART_ENTRY_OFFSET="2048" PART_ENTRY_SIZE="290816" PART_ENTRY_DISK="8:0"

dorileo, 2019-10-02:
hi @adriank41 could you now run: `CBM_DEBUG=1 clr-boot-manager update`?

adriank41, 2019-10-02:
INFO] cbm (../src/bootman/bootman.c:L706): Current running kernel: 5.2.15-834.native
[INFO] cbm (../src/bootman/sysconfig.c:L179): Discovered UEFI ESP: /dev/disk/by-partuuid/d2633b4c-395e-419b-b123-56c0029a5288
[INFO] cbm (../src/bootman/sysconfig.c:L254): Fully resolved boot device: /dev/sda1
[DEBUG] cbm (../src/lib/probe.c:L261): Root device exists on device-mapper configuration
[DEBUG] cbm (../src/bootman/bootman.c:L120): shim-systemd caps: 0x26, wanted: 0x26
[DEBUG] cbm (../src/bootman/bootman.c:L135): UEFI boot now selected (shim-systemd)
[INFO] cbm (../src/bootman/bootman.c:L431): Checking for mounted boot dir
[DEBUG] cbm (../src/lib/files.c:L577): Directory /boot is not empty, found: initramfs-5.1.7-777.native.img
[INFO] cbm (../src/bootman/bootman.c:L439): boot_dir is already mounted: /boot
[DEBUG] cbm (../src/bootman/update.c:L164): Now beginning update_native
[DEBUG] cbm (../src/lib/cmdline.c:L361): Removing cmdline using file: ///etc/kernel/cmdline-removal.d/quiet.conf
[DEBUG] cbm (../src/lib/cmdline.c:L361): Removing cmdline using file: ///etc/kernel/cmdline-removal.d/quiet.conf
[DEBUG] cbm (../src/bootman/update.c:L173): update_native: 2 available kernels
[DEBUG] cbm (../src/bootman/update.c:L191): update_native: Running kernel is (native) ///usr/lib/kernel/org.clearlinux.native.5.2.15-834
[ERROR] cbm (../src/lib/bootvar.c:L319): blkid_probe_get_partitions() error
[ERROR] cbm (../src/lib/bootvar.c:L319): blkid_probe_get_partitions() error
[ERROR] cbm (../src/bootloaders/shim-systemd.c:L284): Cannot create EFI variable (boot entry)
[ERROR] cbm (../src/bootloaders/shim-systemd.c:L285): Please manually update your bios to add a boot entry for Clear Linux
[SUCCESS] cbm (../src/bootman/update.c:L205): update_native: Bootloader updated
[SUCCESS] cbm (../src/bootman/update.c:L220): update_native: Repaired running kernel ///usr/lib/kernel/org.clearlinux.native.5.2.15-834
[DEBUG] cbm (../src/bootman/update.c:L230): update_native: Checking kernels for type native
[INFO] cbm (../src/bootman/update.c:L243): update_native: Default kernel for type native is ///usr/lib/kernel/org.clearlinux.native.5.3.1-843
[SUCCESS] cbm (../src/bootman/update.c:L255): update_native: Installed tip for native: ///usr/lib/kernel/org.clearlinux.native.5.3.1-843
[SUCCESS] cbm (../src/bootman/update.c:L269): update_native: Installed last_good kernel (native) (///usr/lib/kernel/org.clearlinux.native.5.2.15-834)
[DEBUG] cbm (../src/bootman/update.c:L280): update_native: Analyzing for type native: ///usr/lib/kernel/org.clearlinux.native.5.3.1-843
[INFO] cbm (../src/bootman/update.c:L288): update_native: not-running: ///usr/lib/kernel/org.clearlinux.native.5.3.1-843
[DEBUG] cbm (../src/bootman/update.c:L291): update_native: Skipping default-native: ///usr/lib/kernel/org.clearlinux.native.5.3.1-843
[DEBUG] cbm (../src/bootman/update.c:L280): update_native: Analyzing for type native: ///usr/lib/kernel/org.clearlinux.native.5.2.15-834
[DEBUG] cbm (../src/bootman/update.c:L285): update_native: Skipping running kernel
[DEBUG] cbm (../src/lib/cmdline.c:L361): Removing cmdline using file: ///etc/kernel/cmdline-removal.d/quiet.conf
[DEBUG] cbm (../src/lib/cmdline.c:L361): Removing cmdline using file: ///etc/kernel/cmdline-removal.d/quiet.conf
[INFO] cbm (../src/bootman/bootman.c:L431): Checking for mounted boot dir
[DEBUG] cbm (../src/lib/files.c:L577): Directory /boot is not empty, found: initramfs-5.1.7-777.native.img
[INFO] cbm (../src/bootman/bootman.c:L439): boot_dir is already mounted: /boot
[SUCCESS] cbm (../src/bootman/update.c:L338): update_native: Default kernel for native is ///usr/lib/kernel/org.clearlinux.native.5.3.1-843
[DEBUG] cbm (../src/bootman/update.c:L353): No kernel removals found

adriank41, 2019-10-10:
very strange,I solved this issue with cleaning  /boot (was not empty)

closed 2019-10-10 by adriank41