clearlinux/clr-boot-manager #180: No boot entry created

OpenSourceAnarchist, 2019-10-01:
I tried installing Clear Linux* using the new live iso (version 31130).
I have a Dell Latitude 5300 and turned off secure boot.
Currently the only entries showing in my BIOS boot list are from my previous Ubuntu installation.
I have had previous issues with updating the BIOS firmware through LVFS but Ubuntu was able to modify the boot entry list. I tried both "ubuntu" and "Linux boot manager" entries but it always boots me to a "no operating installed" screen.
If it means anything, I've tried the installation twice (both said it was successful) and clr-boot-manager outputs (when running from the live environment):
`lrlinux@clr-live~ $ sudo clr-boot-manager update
[ERROR] cbm (../src/lib/system_stub.c:L31): Invalid block device: 0:26
Out of memory
[FATAL] cbm (../src/bootman/bootman.c:L449): Cannot determine boot device`
Any help would be appreciated. I can provide info through the live iso I still have on my USB flash drive.

dorileo, 2019-10-02:
hi @OpenSourceAnarchist can you send me the `clr-installer.log`(it should be in `/root/`)?
Can you run `blkid` and send me the output? Since now you can boot Clear Linux, could you also run `$ CBM_DEBUG=1  clr-boot-manager update`? if it fails, please send me the output.

OpenSourceAnarchist, 2019-10-02:
@dorileo 

[clr-installer.log](https://github.com/clearlinux/clr-boot-manager/files/3683871/clr-installer.log)

Output of `sudo blkid`:
`/dev/nvme0n1: PTUUID="aa371f71-8cca-4a8e-b44f-e3a6eced012b" PTTYPE="gpt"
/dev/nvme0n1p1: LABEL_FATBOOT="boot" LABEL="boot" UUID="0C1A-EC55" TYPE="vfat" PARTLABEL="EFI" PARTUUID="c2b44353-7077-48f3-a557-5ed29cefd2d0"
/dev/nvme0n1p2: LABEL="swap" UUID="b6cb8084-4c9e-4b41-940d-a9374d1638c1" TYPE="swap" PARTLABEL="linux-swap" PARTUUID="a9ee59ee-7c7a-44de-a59b-2c3b8885fa37"
/dev/nvme0n1p3: LABEL="root" UUID="de78b48e-3e59-4d8b-a4ce-38d8c8af66a9" TYPE="ext4" PARTLABEL="/" PARTUUID="440f14b7-9436-4001-80b5-867e3b839f5b"`

`clr-boot-manager update` ran successfully, but here is the output anyway:
`[INFO] cbm (../src/bootman/bootman.c:L706): Current running kernel: 5.3.1-843.native
[INFO] cbm (../src/bootman/sysconfig.c:L179): Discovered UEFI ESP: /dev/disk/by-partuuid/c2b44353-7077-48f3-a557-5ed29cefd2d0
[INFO] cbm (../src/bootman/sysconfig.c:L254): Fully resolved boot device: /dev/nvme0n1p1
[DEBUG] cbm (../src/bootman/bootman.c:L120): shim-systemd caps: 0x26, wanted: 0x26
[DEBUG] cbm (../src/bootman/bootman.c:L135): UEFI boot now selected (shim-systemd)
[INFO] cbm (../src/bootman/bootman.c:L431): Checking for mounted boot dir
[DEBUG] cbm (../src/lib/files.c:L577): Directory /boot is not empty, found: loader
[INFO] cbm (../src/bootman/bootman.c:L439): boot_dir is already mounted: /boot
[DEBUG] cbm (../src/bootman/update.c:L164): Now beginning update_native
[DEBUG] cbm (../src/bootman/update.c:L173): update_native: 1 available kernels
[DEBUG] cbm (../src/bootman/update.c:L191): update_native: Running kernel is (native) ///usr/lib/kernel/org.clearlinux.native.5.3.1-843
[SUCCESS] cbm (../src/bootman/update.c:L205): update_native: Bootloader updated
[SUCCESS] cbm (../src/bootman/update.c:L220): update_native: Repaired running kernel ///usr/lib/kernel/org.clearlinux.native.5.3.1-843
[DEBUG] cbm (../src/bootman/update.c:L230): update_native: Checking kernels for type native
[INFO] cbm (../src/bootman/update.c:L243): update_native: Default kernel for type native is ///usr/lib/kernel/org.clearlinux.native.5.3.1-843
[SUCCESS] cbm (../src/bootman/update.c:L255): update_native: Installed tip for native: ///usr/lib/kernel/org.clearlinux.native.5.3.1-843
[SUCCESS] cbm (../src/bootman/update.c:L269): update_native: Installed last_good kernel (native) (///usr/lib/kernel/org.clearlinux.native.5.3.1-843)
[DEBUG] cbm (../src/bootman/update.c:L280): update_native: Analyzing for type native: ///usr/lib/kernel/org.clearlinux.native.5.3.1-843
[DEBUG] cbm (../src/bootman/update.c:L285): update_native: Skipping running kernel
[INFO] cbm (../src/bootman/bootman.c:L431): Checking for mounted boot dir
[DEBUG] cbm (../src/lib/files.c:L577): Directory /boot is not empty, found: loader
[INFO] cbm (../src/bootman/bootman.c:L439): boot_dir is already mounted: /boot
[SUCCESS] cbm (../src/bootman/update.c:L338): update_native: Default kernel for native is ///usr/lib/kernel/org.clearlinux.native.5.3.1-843
[DEBUG] cbm (../src/bootman/update.c:L353): No kernel removals found`

I think this is a peculiarity of Dell Latitude firmware because there is a note on the Arch Linux Wiki that reads: 
> Note: The firmware interface of Dell Latitude series provides everything you need to setup EFI boot but the EFI Shell won't be able to write to the computer's ROM.

Perhaps this is not an issue with Clear Linux at all. Just to be certain, if I manually create a boot entry in my UEFI firmware to EFI/BOOT/BOOTX64.EFI, it should just work even after updating the kernel?

mikoim, 2019-11-01:
After clean installing Clear Linux, its boot is successfully but ``clr-boot-manager update`` always failed.
```
$ sudo clr-boot-manager update
[ERROR] cbm (../src/lib/bootvar.c:L412): efi_generate_file_device_path_from_esp() failed: No such file or directory
[ERROR] cbm (../src/lib/bootvar.c:L412): efi_generate_file_device_path_from_esp() failed: No such file or directory
[ERROR] cbm (../src/bootloaders/shim-systemd.c:L284): Cannot create EFI variable (boot entry)
[ERROR] cbm (../src/bootloaders/shim-systemd.c:L285): Please manually update your bios to add a boot entry for Clear Linux
```

In installer environment, it seems works.
```
2019/11/01 14:01:57 [DBG] [INFO] cbm (../src/bootman/bootman.c:L706): Current running kernel: 5.3.8-854.native
2019/11/01 14:01:57 [DBG] [DEBUG] cbm (../src/lib/probe.c:L261): Root device exists on device-mapper configuration
2019/11/01 14:01:57 [DBG] [DEBUG] cbm (../src/bootman/bootman.c:L120): shim-systemd caps: 0x26, wanted: 0x00
2019/11/01 14:01:57 [DBG] [DEBUG] cbm (../src/bootman/bootman.c:L137): Legacy boot now selected (shim-systemd)
2019/11/01 14:01:57 [DBG] [DEBUG] cbm (../src/bootman/update.c:L40): Skipping to image-update
2019/11/01 14:01:57 [DBG] [DEBUG] cbm (../src/bootman/update.c:L76): Now beginning update_image
2019/11/01 14:01:57 [DBG] [DEBUG] cbm (../src/bootman/update.c:L85): update_image: 1 available kernels
2019/11/01 14:01:57 [DBG] [INFO] cbm (../src/bootman/update.c:L111): update_image: Attempting bootloader update
2019/11/01 14:01:57 [DBG] [SUCCESS] cbm (../src/bootman/update.c:L114): update_image: Bootloader update successful
2019/11/01 14:01:57 [DBG] [DEBUG] cbm (../src/bootman/update.c:L125): update_image: Attempting install of /tmp/install-231177564//usr/lib/kernel/org.clearlinux.native.5.3.8-854
2019/11/01 14:01:57 [DBG] [SUCCESS] cbm (../src/bootman/update.c:L130): update_image: Successfully installed /tmp/install-231177564//usr/lib/kernel/org.clearlinux.native.5.3.8-854
2019/11/01 14:01:57 [DBG] [DEBUG] cbm (../src/bootman/update.c:L135): update_image: Setting default_kernel to /tmp/install-231177564//usr/lib/kernel/org.clearlinux.native.5.3.8-854
2019/11/01 14:01:57 [DBG] [INFO] cbm (../src/bootman/bootman.c:L431): Checking for mounted boot dir
2019/11/01 14:01:57 [DBG] [INFO] cbm (../src/bootman/bootman.c:L439): boot_dir is already mounted: /tmp/install-231177564/boot
2019/11/01 14:01:57 [DBG] [SUCCESS] cbm (../src/bootman/update.c:L140): update_image: Default kernel is now /tmp/install-231177564//usr/lib/kernel/org.clearlinux.native.5.3.8-854
```

dorileo, 2020-01-31:
This issue seems to be the same as #208

dorileo, 2020-02-22:
We've updated efivar package some time ago, could you please try it again and see if it works? Thanks..