clearlinux/clr-boot-manager #182: repair is broken on encrypted drive (with btrfs root)

ser, 2019-10-07:
**Describe the bug**
```cryptsetup luksOpen /dev/drive drive```
```mount /dev/mapper/drive /mnt```
```swupd repair --picky --root=/mnt```

gives

```
Calling post-update helper scripts
[ERROR] cbm (../src/lib/files.c:L165): Invalid block device: /mnt
[ERROR] cbm (../src/lib/system_stub.c:L31): Invalid block device: 0:58
Out of memory
[FATAL] cbm (../src/bootman/sysconfig.c:L275): sysconfig insane: Missing root device
[FATAL] cbm (../src/bootman/update.c:L127): Cannot install kernel /mnt//usr/lib/kernel/org.clearlinux.native.5.3.2-844

Repair successful
```

**To Reproduce**
Install clear linux with encryption, ```btrfs-convert /dev/drive``` from installer media, system does not boot, try to fix with the above command

**Expected behavior**
fixing boot by checking that filesystem ID has changed during btrfs convertion

**Environment (please complete the following information):**
swupd-client 3.22.3
Installed version: 31140

seanvk, 2019-10-16:
CBM doesn't support btrfs to my knowledge.  Which is kind of unfortunate but CBM has other boot/security roles beyond just supporting systemd-boot.   So although on other distributions you can freely use btrfs with systemd-boot, CBM cannot.

ahkok, 2019-10-16:
> CBM doesn't support btrfs to my knowledge

That would IMHO be a bug. One that needs fixing.

ahkok, 2019-10-16:
Note that `btrfs-convert` isn't the same as making a new btrfs filesystem. I would NOT recommend btrfs-convert. It's method includes rollback support to ext4, and is likely therefore confusing programs that look at the block content. We first want to assure that filesystems created with `mkfs.btrfs` work anyway.

ser, 2019-10-16:
It would be perfect if installer supports btrfs, but it does not.

I have successfully converted and run clear on btrfs again, but had to manually update a boot text file with a new filesystem ID.