clearlinux/clr-boot-manager #194: swupd bundle-add: Allow install of loader entries and EFI folders regardless of running kernel

seanvk, 2019-11-25:
When running Clear-Linux with a custom kernel which happens to not match the current default entry, adding a new kernel bundle through swupd bundle-add results in a failure to install the new bundle's loader configuration and kernel in the boot mount.

**bundle-add error**
----------
svkelley@ClearLinuxSNR~ $ sudo swupd bundle-add kernel-lts2018-preempt-rt
Loading required manifests...
Downloading packs (35.11 MB) for:
 - kernel-lts2018-preempt-rt
 [100%]

Finishing packs extraction...
Validate downloaded files
 [100%]

No extra files need to be downloaded
 [100%]

Installing files...
 [100%]

Calling post-update helper scripts
External command: none
External command: [ERROR] cbm (../src/bootman/bootman.c:L702): Failed to parse given uname release: 4.19.82-rt30
External command: [WARNING] cbm (../src/bootman/bootman.c:L72): Unable to parse the currently running kernel: 4.19.82-rt30
External command: [ERROR] cbm (../src/lib/bootvar.c:L129): efi_get_next_variable_name() failed: Function not implemented
External command: [FATAL] cbm (../src/bootman/bootman.c:L143): Cannot initialise bootloader shim-systemd
External command: mcelog.service: restarted (the binary was updated)
Successfully installed 1 bundle
----------

After mounting boot mount to /boot and looking in entries and EFI, you can see the RT kernel's files were not populated:

**Entries**
svkelley@ClearLinuxSNR/boot/loader/entries $ ls
Clear-linux-native-5.3.11-869.conf  Clear-linux-native-5.3.12-871.conf

**EFI**
svkelley@ClearLinuxSNR/boot/EFI $ cd org.clearlinux/
svkelley@ClearLinuxSNR/boot/EFI/org.clearlinux $ ls
bootloaderx64.efi                   kernel-org.clearlinux.native.5.3.11-869
bzImage                             kernel-org.clearlinux.native.5.3.12-871
freestanding-00-intel-ucode.cpio    loaderx64.efi
freestanding-i915-firmware.cpio.xz

My suggestion is that it should allow the population of the added bundle files regardless of the fact you are running a custom kernel.

Thanks,

Sean

dorileo, 2019-12-11:
We've seen similar issues with other kernel naming schemes, we added custom code to deal with that. I might come up with a more generic solution to that.

seanvk, 2019-12-12:
Sounds good.  I also had an idea for a future long term enhancement would also allow you to manage both custom and Clear kernels with clr-boot-manager.  But I will create a separate ticket for that.