clearlinux/clr-boot-manager #199: failed to create a new libblkid probe

thiagomacieira, 2019-12-13:
```
 # clr-boot-manager list-kernels
[ERROR] cbm (../src/bootman/sysconfig.c:L97): /dev/disk/by-partuuid: failed to create a new libblkid probe
# ls -l /dev/disk/by-partuuid/
total 0
lrwxrwxrwx 1 root root 10 Dec 11 19:07 46e0b3f4-7666-8949-80f8-285894f4fe76 -> ../../sda2
lrwxrwxrwx 1 root root 10 Dec 11 19:07 ddd66ab5-8807-8946-bb99-bd5beb525414 -> ../../sda1
 # ls -l /sys/firmware/efi/efivars/LoaderDevicePartUUID*
-rw-r--r-- 1 root root 32 Dec 11 19:07 '/sys/firmware/efi/efivars/LoaderDevicePartUUID$ASN-4a67b082-0a4c-41cf-b6c7-440b29bb8c4f'
-rw-r--r-- 1 root root 78 Dec 11 19:07  /sys/firmware/efi/efivars/LoaderDevicePartUUID-4a67b082-0a4c-41cf-b6c7-440b29bb8c4f
```

Bootctl:
```
# bootctl is-installed
yes
# bootctl status
System:
     Firmware: UEFI 2.60 (Parallels EFI x64 14.1.2-45479 14.45479)
  Secure Boot: disabled
   Setup Mode: setup

Current Boot Loader:
      Product: systemd-boot 242
     Features: ✓ Boot counting
               ✓ Menu timeout control
               ✓ One-shot menu timeout control
               ✓ Default entry control
               ✓ One-shot entry control
               ✗ Support for XBOOTLDR partition
               ✗ Support for passing random seed to OS
          ESP: /dev/disk/by-partuuid/ddd66ab5-8807-8946-bb99-bd5beb525414
         File: └─/EFI/org.clearlinux/bootloaderx64.efi

Random Seed:
 Passed to OS: no
 System Token: not set
       Exists: no

Available Boot Loaders on ESP:
          ESP: /efi (/dev/disk/by-partuuid/ddd66ab5-8807-8946-bb99-bd5beb525414)
         File: └─/EFI/systemd/systemd-bootx64.efi (systemd-boot 242)
         File: └─/EFI/BOOT/bootx64.efi (systemd-boot 242)

Boot Loaders Listed in EFI Variables:
        Title: Linux bootloader
           ID: 0x0006
       Status: active, boot-order
    Partition: /dev/disk/by-partuuid/ddd66ab5-8807-8946-bb99-bd5beb525414
         File: └─/EFI/org.clearlinux/bootloaderx64.efi

Boot Loader Entries:
        $BOOT: /efi (/dev/disk/by-partuuid/ddd66ab5-8807-8946-bb99-bd5beb525414)

Default Boot Loader Entry:
        title: Clear Linux OS (Clear-linux-native-5.1.14-788)
           id: Clear-linux-native-5.1.14-788
       source: /efi/loader/entries/Clear-linux-native-5.1.14-788.conf
        linux: /EFI/org.clearlinux/kernel-org.clearlinux.native.5.1.14-788
       initrd: /EFI/org.clearlinux/freestanding-00-intel-ucode.cpio
               /EFI/org.clearlinux/freestanding-i915-firmware.cpio.xz
      options: root=PARTUUID=46e0b3f4-7666-8949-80f8-285894f4fe76 quiet console=tty0 console=ttyS0,115200n8 init=/usr/bin/initra-desktop  initc>
```

dorileo, 2019-12-13:
@thiagomacieira could you please send me the output of: ```lsblk --output-all --json```

thiagomacieira, 2019-12-13:
```json
{
   "blockdevices": [
      {"name":"sda", "kname":"sda", "path":"/dev/sda", "maj:min":"8:0", "fsavail":null, "fssize":null, "fstype":null, "fsused":null, "fsuse%":null, "mountpoint":null, "label":null, "uuid":null, "ptuuid":"b7b0df4f-bc21-5a48-94fd-b3e4e1d4f1a5", "pttype":"gpt", "parttype":null, "partlabel":null, "partuuid":null, "partflags":null, "ra":1024, "ro":false, "rm":false, "hotplug":false, "model":"harddisk", "serial":"MMPAKDTAF19MQ0AM7643", "size":"64.3G", "state":"running", "owner":"root", "group":"disk", "mode":"brw-rw----", "alignment":0, "min-io":512, "opt-io":0, "phy-sec":512, "log-sec":512, "rota":false, "sched":"bfq", "rq-size":1024, "type":"disk", "disc-aln":0, "disc-gran":"512B", "disc-max":"2G", "disc-zero":false, "wsame":"0B", "wwn":null, "rand":true, "pkname":null, "hctl":"0:0:0:0", "tran":"sata", "subsystems":"block:scsi:pci", "rev":"A204", "vendor":"ATA     ", "zoned":"none",
         "children": [
            {"name":"sda1", "kname":"sda1", "path":"/dev/sda1", "maj:min":"8:1", "fsavail":null, "fssize":null, "fstype":"vfat", "fsused":null, "fsuse%":null, "mountpoint":null, "label":"EFI", "uuid":"8A37-2B44", "ptuuid":"b7b0df4f-bc21-5a48-94fd-b3e4e1d4f1a5", "pttype":"gpt", "parttype":"c12a7328-f81f-11d2-ba4b-00a0c93ec93b", "partlabel":"ESP System Partition", "partuuid":"ddd66ab5-8807-8946-bb99-bd5beb525414", "partflags":null, "ra":1024, "ro":false, "rm":false, "hotplug":false, "model":null, "serial":null, "size":"256M", "state":null, "owner":"root", "group":"disk", "mode":"brw-rw----", "alignment":0, "min-io":512, "opt-io":0, "phy-sec":512, "log-sec":512, "rota":false, "sched":"bfq", "rq-size":1024, "type":"part", "disc-aln":0, "disc-gran":"512B", "disc-max":"2G", "disc-zero":false, "wsame":"0B", "wwn":null, "rand":true, "pkname":"sda", "hctl":null, "tran":null, "subsystems":"block:scsi:pci", "rev":null, "vendor":null, "zoned":"none"},
            {"name":"sda2", "kname":"sda2", "path":"/dev/sda2", "maj:min":"8:2", "fsavail":"53G", "fssize":"62.5G", "fstype":"ext4", "fsused":"6.2G", "fsuse%":"10%", "mountpoint":"/", "label":"root", "uuid":"b5fa526f-5328-4ecc-aa71-40b824230139", "ptuuid":"b7b0df4f-bc21-5a48-94fd-b3e4e1d4f1a5", "pttype":"gpt", "parttype":"4f68bce3-e8cd-4db1-96e7-fbcaf984b709", "partlabel":"Root Partition", "partuuid":"46e0b3f4-7666-8949-80f8-285894f4fe76", "partflags":null, "ra":1024, "ro":false, "rm":false, "hotplug":false, "model":null, "serial":null, "size":"64G", "state":null, "owner":"root", "group":"disk", "mode":"brw-rw----", "alignment":0, "min-io":512, "opt-io":0, "phy-sec":512, "log-sec":512, "rota":false, "sched":"bfq", "rq-size":1024, "type":"part", "disc-aln":0, "disc-gran":"512B", "disc-max":"2G", "disc-zero":false, "wsame":"0B", "wwn":null, "rand":true, "pkname":"sda", "hctl":null, "tran":null, "subsystems":"block:scsi:pci", "rev":null, "vendor":null, "zoned":"none"}
         ]
      }
   ]
}
```

dorileo, 2019-12-14:
@thiagomacieira can you run it with debug? i.e ```CBM_DEBUG=1 clr-boot-manager list-kernels```

thiagomacieira, 2019-12-14:
Sure:
```
 # CBM_DEBUG=1 clr-boot-manager list-kernels
[INFO] cbm (../src/bootman/bootman.c:L706): Current running kernel: 5.1.14-788.native
[INFO] cbm (../src/bootman/sysconfig.c:L179): Discovered UEFI ESP: /dev/disk/by-partuuid/
[INFO] cbm (../src/bootman/sysconfig.c:L256): Fully resolved boot device: /dev/disk/by-partuuid
[ERROR] cbm (../src/bootman/sysconfig.c:L97): /dev/disk/by-partuuid: failed to create a new libblkid probe
```

thiagomacieira, 2019-12-14:
As explained on IRC, strace shows that cbm is opening and reading file `/sys/firmware/efi/efivars/LoaderDevicePartUUID$ASN-4a67b082-0a4c-41cf-b6c7-440b29bb8c4f`. That file does not contain a UUID.

PR #209 (2020-01-28): get_boot_device: fix efi loader path matching
We've found some cases where the path to LoaderDevicePartUUID will have
some "virtual" entries containing some "garbage" in the path. This patch
makes sure we always match only the right formatted ones.

Fixes: #199
closed 2020-01-28 by bryteise