clearlinux/clr-boot-manager #205: clr-boot-manager doesn't remove old Clear kernels when booted in a custom kernel

cmarcelo, 2020-01-14:
When `clr-boot-manager update` is called (manually or as part of `swupd update`) while the machine is running a custom kernel (not the ones provided by Clear), it seems to exit before it can clear up old kernels. Would be nice if it still could remove older kernels.

By default Clear keeps the running and the latest. Perhaps in this case that the running kernel is not from Clear, it could keep the two latest ones provided by Clear.

Note: I'm considering cbm will **never** remove a custom kernel -- which I think is the correct behavior.

ahkok, 2020-01-14:
transferred the issue - highly specific to CBM.

dorileo, 2020-01-22:
Thank you @cmarcelo cbm shouldn't fail in that case, I'll take a look at that. Thanks.

tuzhms, 2021-05-24:
I have the same problem. I delete old kernels manually when filling the `/boot` partition

closed 2022-10-05 by cmarcelo