clearlinux/clr-boot-manager #208: EFI Errors after update/repair

0xA1B2, 2020-01-24:
Calling post-update helper scripts
External command: none
External command: [ERROR] cbm (../src/lib/bootvar.c:L412): efi_generate_file_device_path_from_esp() failed: No such file or directory
External command: [ERROR] cbm (../src/lib/bootvar.c:L412): efi_generate_file_device_path_from_esp() failed: No such file or directory
External command: [ERROR] cbm (../src/bootloaders/shim-systemd.c:L284): Cannot create EFI variable (boot entry)
External command: [ERROR] cbm (../src/bootloaders/shim-systemd.c:L285): Please manually update your bios to add a boot entry for Clear Linux

This happens after swupd update or repair

Same on Lenovo T520 and Intel Compute Stick STK1AW32SC, both with latest BIOS.

Seems to be related to older hardware ? Maybe the errors could be suppressed if the hardware is too old.

dorileo, 2020-01-30:
The root cause of this error lies with efivar, the newest efivar release (37) fixes the issue. I'm working to update it.

dorileo, 2020-01-30:
Just pushed a new version of efivar, might land on Clear Linux any time soon.

generator75, 2020-02-03:
The error msgs related to efivar are gone, but no update occurs :
DEBUG] cbm (../src/cli/cli.c:L142): No such file: //etc/kernel/update_efi_vars
[INFO] cbm (../src/bootman/bootman.c:L734): Current running kernel: 5.4.2-875.native
[INFO] cbm (../src/bootman/sysconfig.c:L179): Discovered UEFI ESP: /dev/disk/by-partuuid/6f22908a-e7af-2f0a-43e9-07730c15275c
[INFO] cbm (../src/bootman/sysconfig.c:L256): Fully resolved boot device: /dev/sdc2
[DEBUG] cbm (../src/bootman/bootman.c:L140): shim-systemd caps: 0x26, wanted: 0x26
[DEBUG] cbm (../src/bootman/bootman.c:L155): UEFI boot now selected (shim-systemd)
[INFO] cbm (../src/bootman/bootman.c:L754): path ///etc/kernel/initrd.d does not exist
[DEBUG] cbm (../src/bootman/bootman.c:L774): Non regular file: ///usr/lib/initrd.d/.., skipping.
[DEBUG] cbm (../src/bootman/bootman.c:L774): Non regular file: ///usr/lib/initrd.d/., skipping.
[INFO] cbm (../src/bootman/bootman.c:L459): Checking for mounted boot dir
[DEBUG] cbm (../src/lib/files.c:L577): Directory /boot is not empty, found: loader
[INFO] cbm (../src/bootman/bootman.c:L467): boot_dir is already mounted: /boot
[DEBUG] cbm (../src/bootman/update.c:L164): Now beginning update_native
[DEBUG] cbm (../src/bootman/update.c:L173): update_native: 2 available kernels
[DEBUG] cbm (../src/bootman/update.c:L191): update_native: Running kernel is (native) ///usr/lib/kernel/org.clearlinux.native.5.4.2-875
[SUCCESS] cbm (../src/bootman/update.c:L205): update_native: Bootloader updated
[SUCCESS] cbm (../src/bootman/update.c:L220): update_native: Repaired running kernel ///usr/lib/kernel/org.clearlinux.native.5.4.2-875
[DEBUG] cbm (../src/bootman/update.c:L230): update_native: Checking kernels for type native
[INFO] cbm (../src/bootman/update.c:L243): update_native: Default kernel for type native is ///usr/lib/kernel/org.clearlinux.native.5.4.16-900
[SUCCESS] cbm (../src/bootman/update.c:L255): update_native: Installed tip for native: ///usr/lib/kernel/org.clearlinux.native.5.4.16-900
[SUCCESS] cbm (../src/bootman/update.c:L269): update_native: Installed last_good kernel (native) (///usr/lib/kernel/org.clearlinux.native.5.4.2-875)
[DEBUG] cbm (../src/bootman/update.c:L280): update_native: Analyzing for type native: ///usr/lib/kernel/org.clearlinux.native.5.4.16-900
[INFO] cbm (../src/bootman/update.c:L288): update_native: not-running: ///usr/lib/kernel/org.clearlinux.native.5.4.16-900
[DEBUG] cbm (../src/bootman/update.c:L291): update_native: Skipping default-native: ///usr/lib/kernel/org.clearlinux.native.5.4.16-900
[DEBUG] cbm (../src/bootman/update.c:L280): update_native: Analyzing for type native: ///usr/lib/kernel/org.clearlinux.native.5.4.2-875
[DEBUG] cbm (../src/bootman/update.c:L285): update_native: Skipping running kernel
[INFO] cbm (../src/bootman/bootman.c:L459): Checking for mounted boot dir
[DEBUG] cbm (../src/lib/files.c:L577): Directory /boot is not empty, found: loader
[INFO] cbm (../src/bootman/bootman.c:L467): boot_dir is already mounted: /boot
[SUCCESS] cbm (../src/bootman/update.c:L338): update_native: Default kernel for native is ///usr/lib/kernel/org.clearlinux.native.5.4.16-900
[DEBUG] cbm (../src/bootman/update.c:L353): No kernel removals found

dorileo, 2020-02-03:
@generator75 what do you expect? I mean, how did you check if your boot entries were updated or not? what's the currently running kernel? what is the default set after the update? can you try umounting `/boot` and the running `clr-boot-manager update` again?

dorileo, 2020-02-03:
@puneetse can you check you hyper-v setup? my setup (after the updates with 32230) is working fine.

puneetse, 2020-02-04:
It is working without the EFI errors. Thanks.

generator75, 2020-02-04:
> @generator75 what do you expect? I mean, how did you check if your boot entries were updated or not? what's the currently running kernel? what is the default set after the update? can you try umounting `/boot` and the running `clr-boot-manager update` again?

Hello! Did it the other way around : /boot was not mounted. However, the folder exists and was poupulated with former entries. 
My previous msg :
_Discovered UEFI ESP: /dev/disk/by-partuuid/6f22908a-e7af-2f0a-43e9-07730c15275c
[INFO] cbm (../src/bootman/sysconfig.c:L256): Fully resolved boot device: /dev/sdc2_
Actually, /sdc2 was not mounted : clr-boot-manager thought it was because /boot existed and was populated.
Did mount /dev/sdc2 to /boot and clr-boot-manager did its updating work.
I was stuck with org.clearlinux.native.5.4.2-875 and now am with org.clearlinux.native.5.4.16-900.
Sorry for the delay : I'm at work and Clear is not my main OS.. yet!

dorileo, 2020-02-04:
@generator75 good to hear, thank you.

bktan8, 2020-02-04:
@dorileo - I can also confirm I'm no longer see these CBM issues on Proxmox and Parallels VMs.

dorileo, 2020-02-04:
since we're all happy with it now I'm closing this issue. Please feel free to reopen it if necessary. Thank you.

closed 2020-02-04 by dorileo

generator75, 2020-02-10:
Last update was a sweet breeze. Issue obviously closed. 
Thks Dorileo!