clearlinux/clr-boot-manager #218: No Booting after install

severnside-droid, 2020-02-20:
I have installed Clear Linux ( I am dual booting) , I followed the instructions " Dual-boot Clear Linux* OS with Any GRUB-based Linux* Distro " and used Method 2: Use GRUB to Boot Clear Linux OS.
On booting and selecting Clear Linux I get this " error’ EFI/org.clearlinux/bootloaderx64.efi’ not found "

bktan8, 2020-02-27:
@severnside-droid - What's the other Linux distro (and version) you're trying to dual-boot Clear Linux with?  I want to see if I can reproduce the issue you're experiencing.

severnside-droid, 2020-02-27:
The other Linux distro is MX 19.1 

[image]
[image]

bktan8, 2020-02-28:
@severnside-droid - I tested MX 19.1 with Clear 32340 using the instruction on https://docs.01.org/clearlinux/latest/tutorials/multi-boot/dual-boot-linux.html# and using Method #2 - Use GRUB to Boot Clear Linux OS.  

It works without any issue.  

Did you set the proper "root" UUID (based on the result of blkid) value for /etc/grub.d/40_custom?

severnside-droid, 2020-02-28:
My last attempt at installing Clear Linux was by safe method on the remaining free space as previous attempts had failed.
Please see output of /etc/grub.d , pointing to device sda/5
it should match with the disk in output attached
[image]

[image]

bktan8, 2020-02-28:
I did a fresh installation of MX and telling the installer to "Auto-install using entire disk", with "Leave free space up to" and left enough space for Clear Linux (minimum of ~22GB).  I told it to install GRUB and set location to "ESP".  After MX was done, I installed Clear Linux using the "Safe installation" method.  Then I add Clear to GRUB's boot menu per the instruction in the document.  I didn't run into any issue.  

Here's what my partitions look like:

``$ sudo blkid
/dev/sda1: UUID="5E05-D9C1" TYPE="vfat" PARTLABEL="ESP" PARTUUID="0d1ca6fb-7380-4bbb-9fa2-fcc4745c8afe"
/dev/sda2: LABEL="rootMX19" UUID="58ccace6-faa1-4c2f-a0df-5dd009abd647" TYPE="ext4" PARTLABEL="primary" PARTUUID="8b1656ba-4c87-46b0-a29f-6db35fd36354"
/dev/sda3: LABEL="swapMX" UUID="8edf0916-5ea0-48d3-9711-67cea5ba3448" TYPE="swap" PARTLABEL="primary" PARTUUID="3b8a1569-6489-4a95-8a1e-8198bc0b649c"
/dev/sda4: LABEL_FATBOOT="boot" LABEL="boot" UUID="8826-15CA" TYPE="vfat" PARTLABEL="EFI" PARTUUID="da9f645a-ca3e-42bd-a4eb-7236b53bb02e"
/dev/sda5: LABEL="root" UUID="081fb5b9-095a-4568-b413-777717882e29" TYPE="ext4" PARTLABEL="/" PARTUUID="44dfcf35-4ca2-43fc-9859-853baee13bfe"
/dev/sdb1: LABEL_FATBOOT="U" LABEL="U" UUID="7B71-11FB" TYPE="vfat"

$ cat /etc/grub.d/40_custom
#!/bin/sh
exec tail -n +3 $0
# This file provides an easy way to add custom menu entries.  Simply type the
# menu entries you want to add after this comment.  Be careful not to change
# the 'exec tail' line above.
menuentry 'Clear Linux OS' {
        search --fs-uuid --no-floppy --set=root 8826-15CA
        chainloader (${root})/EFI/org.clearlinux/bootloaderx64.efi
}

$ sudo update-grub
Generating grub configuration file ...
Found theme: /boot/grub/themes/linen/theme.txt
Found linux image: /boot/vmlinuz-4.19.0-6-amd64
Found initrd image: /boot/initrd.img-4.19.0-6-amd64
Found mtest-64.efi image: /boot/uefi-mt/mtest-64.efi
Found Clear Linux OS on /dev/sda5
Adding boot menu entry for EFI firmware configuration
done
``

Is doing a fresh installation of MX and then Clear an option for?  I see a lot of different partitions from your screenshot.  Might be easier to start fresh, if possible.

severnside-droid, 2020-02-28:
Thanks for all your help. Maybe in the future I will either put another disk in the system and just use it for Clear Linux.