clearlinux/clr-boot-manager #220: Support to create images bootable via SBL

thomasmerx, 2020-03-10:
Intel is providing the [Slim Bootloader](https://www.intel.com/sbl) as a fast and lightweight open-source reference boot loader. This is commonly used in IOT devices like e.g. infotainment systems in the car. 
I would like to see the clr-installer directly supporting the creation of images that are bootable via SBL.

ahkok, 2020-03-10:
Migrated away from `clr-installer` because that just asks `clr-boot-manager` to manage bootloaders.

ahkok, 2020-03-10:
The ClearLinux team will likely never use `sbl`. We'd need someone to implement and validate this that can vouch for the correct functionality. Can you?

thomasmerx, 2020-03-11:
In the meantime we decided not to use ClearLinux due to limited support for embedded use-cases, so we currently are not interested to add this support and will continue using Yocto.

closed 2020-03-11 by thomasmerx