clearlinux/clr-boot-manager #225: Suggestion: CBM - Give each kernel a unique number/code so it's easier to select instead of typing out full name

bktan8, 2020-03-23:
It would be nice to put  a unique number/code in front of each kernel when I do a "list".
And then when I do a "set", I can just use the number/code instead of having to type up the full name.
Less chance for mistakes that way.

ahkok, 2020-03-23:
The list could change, making the numbers invalid. But yes, in a console, where you can't copy and paste, this would be a problem.

`tab` completion may be a good solution here.

bryteise, 2020-03-23:
`set` having tab completion would be reasonable.

dorileo, 2020-03-23:
we could hash the kernel name and support a short hash.

puneetse, 2020-03-23:
The output of `efibootmgr` comes to mind. Maybe that style?

dorileo, 2020-03-23:
@puneetse that's exactly what @ahkok is trying to avoid. I agree with him, the list may change between calling list and set kernel.

bryteise, 2020-03-23:
Tab completion works for me in zsh btw. @lebensterben submitted a PR so I think this is fine as is.