clearlinux/clr-boot-manager #226: CBM removed my default "older" kernel

bktan8, 2020-03-23:
For some reason, CBM removed an older kernel that I had set as my default kernel.  See below.  

```
clear@clr~ $ sudo clr-boot-manager list-kernels
Password: 
* org.clearlinux.native.5.5.10-921
  org.clearlinux.native.5.5.8-917

clear@clr~ $ sudo clr-boot-manager set-kernel org.clearlinux.native.5.5.8-917

clear@clr~ $ sudo clr-boot-manager set-timeout 25
New timeout value is: 25

clear@clr~ $ sudo reboot
packet_write_wait: Connection to UNKNOWN port 65535: Broken pipe

bktan1@bktan1 ~ % ssh clear@...
Password: 

clear@clr-mx~ $ sudo clr-boot-manager list-kernels
Password: 
* org.clearlinux.native.5.5.10-921

clear@clr-mx~ $ sudo mount /dev/sda4 /mnt

clear@clr-mx~ $ tree /mnt
/mnt
├── EFI
│   ├── BOOT
│   │   └── BOOTX64.EFI
│   └── org.clearlinux
│       ├── bootloaderx64.efi
│       ├── freestanding-00-intel-ucode.cpio
│       ├── freestanding-i915-firmware.cpio.xz
│       ├── kernel-org.clearlinux.native.5.5.10-921
│       └── loaderx64.efi
└── loader
    ├── entries
    │   └── Clear-linux-native-5.5.10-921.conf
    └── loader.conf

5 directories, 8 files

```

bryteise, 2020-03-23:
This is working as intended. Default kernels aren't safe from cleanup.

closed 2020-03-23 by bryteise

dorileo, 2020-03-23:
Is there any reason not to? if the user wants to keep a kernel we should allow it, otherwise we force the user to [re]build a kernel to use it, what's the point? do we allow the user to stick to a specific kernel or not?

bryteise, 2020-03-23:
We don't, just like we don't let users stick to a specific bundle version. set is for switching kernel types not keeping an old unsupported kernel version.

ahkok, 2020-03-23:
I ... really don't like this amount of restrictions. I have been, many times, in a case where having a 2+ year old kernel ready to boot, would have saved my life. Having the ability to recover from a bad kernel at all times is a feature that is totally worth it and should be part of any serious backup strategy.

This isn't like bundle versions, at all.

bryteise, 2020-03-23:
We have the ability to recover a bad kernel, that's what CBM does. It doesn't need folks to manually keep old kernels we publish around.