clearlinux/clr-boot-manager #227: Suggestion: Need ability to "whitelist" kernels so CBM doesn't automatically remove them

bktan8, 2020-03-23:
I have seen CBM automatically removed older kernels without the user knowing.  
We need ability to whitelist older kernels so CBM doesn't do that.

bryteise, 2020-03-23:
I don't want to see folks using CBM to keep older kernels around. If users really want this they can build and install an older version themselves or rename.

closed 2020-03-23 by bryteise

dorileo, 2020-03-23:
similarly to what I said on #226 I understand that it would benefit users, the user has a kernel that he/she wants to keep using, we don't allow but tell the user to manually build and install it, what's the point? the kernel is just there, why not allow the user to "keep" it? instead we force the user to build something that was just right there.

bryteise, 2020-03-23:
As I said on #226 keeping old kernels isn't a supported workflow, just like keeping old bundles.