clearlinux/clr-boot-manager #248: Segfault when running update on CLR 36110

MartB, 2022-03-30:
**Describe the bug**
Segmentation fault

**To Reproduce**
Run clr-bootmanager update

**Environment (please complete the following information):**
 - Clear Linux OS Version: 36110
 - clr-boot-manager Version: 3.2.9
 - Platform: x86 legacy MBR VM (lts2020 kernel)

** Block Devices:**
 - List of Block Devices: [lsblk -a --json]
```
{
   "blockdevices": [
      {
         "name": "loop0",
         "maj:min": "7:0",
         "rm": false,
         "size": "0B",
         "ro": false,
         "type": "loop",
         "mountpoints": [
             null
         ]
      },{
         "name": "loop1",
         "maj:min": "7:1",
         "rm": false,
         "size": "0B",
         "ro": false,
         "type": "loop",
         "mountpoints": [
             null
         ]
      },{
         "name": "loop2",
         "maj:min": "7:2",
         "rm": false,
         "size": "0B",
         "ro": false,
         "type": "loop",
         "mountpoints": [
             null
         ]
      },{
         "name": "loop3",
         "maj:min": "7:3",
         "rm": false,
         "size": "0B",
         "ro": false,
         "type": "loop",
         "mountpoints": [
             null
         ]
      },{
         "name": "loop4",
         "maj:min": "7:4",
         "rm": false,
         "size": "0B",
         "ro": false,
         "type": "loop",
         "mountpoints": [
             null
         ]
      },{
         "name": "loop5",
         "maj:min": "7:5",
         "rm": false,
         "size": "0B",
         "ro": false,
         "type": "loop",
         "mountpoints": [
             null
         ]
      },{
         "name": "loop6",
         "maj:min": "7:6",
         "rm": false,
         "size": "0B",
         "ro": false,
         "type": "loop",
         "mountpoints": [
             null
         ]
      },{
         "name": "loop7",
         "maj:min": "7:7",
         "rm": false,
         "size": "0B",
         "ro": false,
         "type": "loop",
         "mountpoints": [
             null
         ]
      },{
         "name": "vda",
         "maj:min": "253:0",
         "rm": false,
         "size": "25G",
         "ro": false,
         "type": "disk",
         "mountpoints": [
             null
         ],
         "children": [
            {
               "name": "vda1",
               "maj:min": "253:1",
               "rm": false,
               "size": "149M",
               "ro": false,
               "type": "part",
               "mountpoints": [
                   "/boot"
               ]
            },{
               "name": "vda2",
               "maj:min": "253:2",
               "rm": false,
               "size": "24,9G",
               "ro": false,
               "type": "part",
               "mountpoints": [
                   "/"
               ]
            }
         ]
      }
   ]
}
```

**Additional context**
```
Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7f84d86 in blkid_do_safeprobe (pr=0x0) at libblkid/src/probe.c:1561
1561	libblkid/src/probe.c: No such file or directory.
(gdb) bt
#0  0x00007ffff7f84d86 in blkid_do_safeprobe (pr=0x0) at libblkid/src/probe.c:1561
#1  0x00005555555616f6 in get_parent_disk_devno (diskdevno=<synthetic pointer>, path=0x555555577520 "/") at ../src/lib/files.c:166
#2  get_parent_disk (path=0x555555577520 "/") at ../src/lib/files.c:208
#3  0x00005555555619f3 in get_legacy_boot_device (path=path@entry=0x555555577520 "/") at ../src/lib/files.c:283
#4  0x000055555555f3d8 in cmb_inspect_root_native (realp=0x555555577520 "/", c=0x5555555786a0) at ../src/bootman/sysconfig.c:183
#5  cbm_inspect_root (path=path@entry=0x55555556e931 "/", image_mode=<optimized out>) at ../src/bootman/sysconfig.c:243
#6  0x000055555555b0c3 in boot_manager_set_prefix (self=self@entry=0x555555577e50, prefix=prefix@entry=0x55555556e931 "/") at ../src/bootman/bootman.c:184
#7  0x000055555555ae02 in cbm_command_update_do (manager=0x555555577e50, root=0x0, forced_image=<optimized out>) at ../src/cli/ops/update.c:77
#8  0x000055555555aedc in cbm_command_update (argc=<optimized out>, argv=<optimized out>) at ../src/cli/ops/update.c:44
#9  0x0000555555559772 in main (argc=0, argv=0x7fffffffc1b8) at ../src/cli/main.c:280
```

quizac-, 2022-05-11:
It actually starts segfaulting from release 36080. I was bisecting it from 35000(default on AWS market) til latest in order to find last stable release.

35000 ok
35800 ok
36000 ok
36050 ok
36060 ok
36070 ok
36080 clr-boot-manager update: segfault
36100 clr-boot-manager update: segfault
36280 clr-boot-manager update: segfault

quizac-, 2022-05-11:
It seems that upgrade process from 36070 to 36080 removed /dev/root and more recent binaries cannot cope with it. I'm using AWS provided image so no extra /boot partition is needed hence creating /dev/root symlink was sufficient to mitigate the issue. From my perspective, it seems to be quite severe bc CL3 would be first choice for those who demand extra kick given by performance optimization and tuning kernel params is a common step to configure isolcpus, etc while this bug prevents CL3 from running at full potential. Thanks @MartB for workaround!

quizac-, 2022-05-12:
Below you can find temporary solution which creates /dev/root at boot:
cat /etc/systemd/system/cbm_fix.service:
`[Unit]
Description=cbm_fix
After=local-fs.target

[Service]
ExecStart=/usr/sbin/cbm_fix
RemainAfterExit=yes
Type=oneshot
User=root
Group=root
StandardOutput=syslog
StandardError=syslog
SyslogIdentifier=cbm_fix

[Install]
WantedBy=multi-user.target
`

/usr/sbin/cbm_fix:
`#!/bin/bash

root_dev=`mount | grep -F ' on / type ' | awk '{ print $1 }'`
[ -n "$root_dev" ] && ln -sf "$root_dev" /dev/root

exit 0
`

closed 2024-11-11 by MartB