clearlinux/clr-boot-manager #255: failed to update - No space left on device

hajes, 2022-12-30:
5 min old fresh install of Clear Linux 37860...trying to update now

/etc/kernel/cmdline.d/mitigations.conf >>
mitigations=off

`sudo clr-boot-manager update`

`[ERROR] cbm (../src/lib/bootvar.c:L372): efi_set_variable() failed: No space left on device`
`[ERROR] cbm (../src/bootloaders/shim-systemd.c:L284): Cannot create EFI variable (boot entry)`
`[ERROR] cbm (../src/bootloaders/shim-systemd.c:L285): Please manually update your bios to add a boot entry for Clear Linux`

**Expected behavior**
no errors as before

**Environment (please complete the following information):**
 - Clear Linux OS Version: 37930
 - clr-boot-manager Version: 3.2.12
 - Platform: Ryzen 9 5950X, ASUS ProArt B550-Creator, Ripjaw 128GB, nVidia Quadro 600 (non EFI, might be problem), Samsung 256GB NVMe

** Block Devices:**
 - List of Block Devices:

{
   "blockdevices": [
      {
         "name": "loop0",
         "maj:min": "7:0",
         "rm": false,
         "size": "0B",
         "ro": false,
         "type": "loop",
         "mountpoints": [
             null
         ]
      },{
         "name": "loop1",
         "maj:min": "7:1",
         "rm": false,
         "size": "0B",
         "ro": false,
         "type": "loop",
         "mountpoints": [
             null
         ]
      },{
         "name": "loop2",
         "maj:min": "7:2",
         "rm": false,
         "size": "0B",
         "ro": false,
         "type": "loop",
         "mountpoints": [
             null
         ]
      },{
         "name": "loop3",
         "maj:min": "7:3",
         "rm": false,
         "size": "0B",
         "ro": false,
         "type": "loop",
         "mountpoints": [
             null
         ]
      },{
         "name": "loop4",
         "maj:min": "7:4",
         "rm": false,
         "size": "0B",
         "ro": false,
         "type": "loop",
         "mountpoints": [
             null
         ]
      },{
         "name": "loop5",
         "maj:min": "7:5",
         "rm": false,
         "size": "0B",
         "ro": false,
         "type": "loop",
         "mountpoints": [
             null
         ]
      },{
         "name": "loop6",
         "maj:min": "7:6",
         "rm": false,
         "size": "0B",
         "ro": false,
         "type": "loop",
         "mountpoints": [
             null
         ]
      },{
         "name": "loop7",
         "maj:min": "7:7",
         "rm": false,
         "size": "0B",
         "ro": false,
         "type": "loop",
         "mountpoints": [
             null
         ]
      },{
         "name": "nvme0n1",
         "maj:min": "259:0",
         "rm": false,
         "size": "1.8T",
         "ro": false,
         "type": "disk",
         "mountpoints": [
             null
         ],
         "children": [
            {
               "name": "nvme0n1p1",
               "maj:min": "259:1",
               "rm": false,
               "size": "1.8T",
               "ro": false,
               "type": "part",
               "mountpoints": [
                   "/c-tmp"
               ]
            }
         ]
      },{
         "name": "nvme1n1",
         "maj:min": "259:2",
         "rm": false,
         "size": "238.5G",
         "ro": false,
         "type": "disk",
         "mountpoints": [
             null
         ],
         "children": [
            {
               "name": "nvme1n1p1",
               "maj:min": "259:3",
               "rm": false,
               "size": "512M",
               "ro": false,
               "type": "part",
               "mountpoints": [
                   null
               ]
            },{
               "name": "nvme1n1p2",
               "maj:min": "259:4",
               "rm": false,
               "size": "7.3G",
               "ro": false,
               "type": "part",
               "mountpoints": [
                   "[SWAP]"
               ]
            },{
               "name": "nvme1n1p3",
               "maj:min": "259:5",
               "rm": false,
               "size": "230.6G",
               "ro": false,
               "type": "part",
               "mountpoints": [
                   "/"
               ]
            }
         ]
      }
   ]
}

**Additional context**
After update & reboot...still same error, but additional kernel parameters are loaded...I guess it works, just with error.

hajes, 2022-12-31:
Tried to install nVidia driver with DKMS...it doesn't update EFI...no boot into new kernel.

bryteise, 2023-01-01:
`efi_set_variable() failed: No space left on device` seems to indicate clr-boot-manager couldn't update your efi variable. Not sure how you can adjust that with your bios but you might be able to. I don't think clr-boot-manager can do anything else here though.

hajes, 2023-01-01:
`/boot` is empty, my Debian with Grub contains kernel images.  Why can it not update? Is it booted/installed in legacy mode?

bryteise, 2023-01-01:
Oh sorry, missed your mention it wasn't EFI though I'm a bit confused about the install as you have nvme1n1p1 which looks like it was setup as an EFI partition. /boot isn't something we keep mounted (clr-boot-manager will mount and unmount it if it wasn't mounted before) so that might be why it is empty.

hajes, 2023-01-01:
it suppose to be EFI. But what installer actually did, no idea. I know MB BIOS whined about non-EFI GPU. My guess is, it is somehow installed in legacy mode.

Sadly, I am unable to boot another EFI GPU...I will have a new GPU in 2 days. Maybe, it resolves the issue. Hopefully, without reinstalling OS again.

bryteise, 2023-01-04:
I really don't understand the idea of a non-EFI GPU (other than perhaps the EFI bios itself can't work with the gpu for some graphical work) the installer seems to have done an EFI install based on the partition so I suspect the partition does have contents.